Atmosphere-NX/Atmosphere #1128: Failed to boot a 4.1.0 partition-based emunand

jejer, 2020-08-17:
## Bug Report

### What's the issue you encountered?
Both hekete and fusee-primary failed to boot a 4.1.0 partition-based emunand emunand.
Got a black screen after Nintendo Switch Logo.
But a file-based emunand boots fine which created from the same sysnand.
**And after reboot to hekete from the file-based emunand 4.1.0, the partition-based emunand 4.1.0 could be booted successfully.**
After upgrade the partition-based emunand to 10.1.0, it boots ok.

### How can the issue be reproduced?

Create a partition and emunand using hekate from sysnand 4.1.0.
This partition-based emunand always fail to boot.
Tested on two different brands of SD cards.

### System Firmware Version

4.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate and fusee-primary
- Official release or unofficial build: 
  - Atmosphère 0.14.1
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: ChoiDujourNX

### Additional context?

RCM Exploitable Switch
Auto RCM disabled
Caffeine used to boot into RCM
incognito applied on sysnand

SciresM, 2020-08-24:
I have also encountered this -- spurious boot issues (and *particularly* long wait times to get to qlaunch display). Launch disp times are shorter and boot hasn't failed for me on 3.x or 6.x+.

It affects 4.x emunand. I've seen it affect 5.x once.

I am reasonably confident it is an emummc issue, and not an atmosphere one. Issue should really be made on the emummc repo.

SciresM, 2020-08-24:
The workaround, as you've noticed, is to upgrade your emunand past 4.x, and past 5.x to be safe.

closed 2020-08-24 by SciresM