Atmosphere-NX/Atmosphere #1132: Daybreak won't work properly.

PaperBro, 2020-08-27:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Decided to update my emuMMC fw version, first I updated hekate and Atmosphere ]</br>
[ Yes, I updated my Atmosphere version to the latest release as of 28/08/2020 ]</br>
[  ]</br>

### How can the issue be reproduced?

[ I followed the update process present on https://nh-server.github.io/switch-guide/extras/updating/    Once I started updating the EmuMMC itself through Daybreak, everything seemed to go smoothly. I had already dumped my firmware version from Sysnand. However, after Daybreak said that the update was complete and I rebooted, I found that my emuMMC fw version had not updated at all. ]</br>
[ ]</br>

### Crash Report

[ N/A ]</br>
[ N/A]</br>
[ N/A ]</br>

### System Firmware Version

10.1.0(On sysNAND)</br>
[ ]</br>
[ ]</br>
[ ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: Official
  - [ Official release version 5.3.2  ]
  - [ N/A ] 
     - [ ]
     - [  ]
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: [ HB App Store, Goldleaf, Daybreak, Edizon, Kosmos Toolbox, Lockpick]

### Additional context?

- Additional info about your environment: N/A
- [ N/A ]
[image]
[image]
[image]

 ]

nwhiteman42, 2020-09-29:
Are you by chance launching the Homebrew Launcher as an applet instead of using app override?

PaperBro, 2020-09-29:
No, I fixed it, turns out the dumped fw was corrupted or smth, I just
re-dumped it and it worked fine.

closed 2020-09-29 by SciresM