Atmosphere-NX/Atmosphere #1135: Error launching games that used to start

Afrodit20, 2020-09-02:
## Bug Report

When I install Untitled goose game, It works ok, but for one reason other games that used to run no longer open </br>

### What's the issue you encountered?

I install untitled Goose game, and after my pokemon shield and other 3 games dont start and say "the program could not be started"</br>

### How can the issue be reproduced?

I dont know, its the first time i encounter this and i odn know how to fix it</br>

### Crash Report

Only says: "the program could not be started"</br>

### System Firmware Version

hbmenu: v3.3.0</br>
nx-hbloader: v2.3.4</br>
Atmosphere: v0.14.1</br>
Nswich: v10.1.0</br>
### Environment?

I instal the untitled goose game and after we cant open the pokemon and other 3 games (Liberated, Little Missfortune, Wonderfull101)</br>

### Additional context?

I don't know how to read the report and it's the first time I get this error and I cannot get it anymore. If someone can help me I would appreciate it very much</br>

Masamune3210, 2020-09-02:
exfat sd card or fat32?

jinkjones, 2020-09-16:
Hello, did you manage to fix this? everytime i try to load a game it doesnt work and has the same error.

jinkjones, 2020-09-16:
Resolved for me, redownload your sig patches from awoo installer.

fincs, 2020-09-16:
Piracy is not supported.

closed 2020-09-16 by fincs