Atmosphere-NX/Atmosphere #1137: Error 0x35F202 (2002-6905) when loading Smash Ultimate mods with ARCropolis

WuBoytH, 2020-09-06:
## Bug Report

### What's the issue you encountered?

I'm making a Smash Ultimate modpack and today, after installing a custom theme, the modpack started throwing error 0x35F202 (2002-6905). I have reinstalled Atmosphere multiple times, but I am unable to load mods without this error popping up now.</br>
As far as I can tell, this is an SD card mounting issue, but it's never happened before I installed that custom theme.</br>

### How can the issue be reproduced?

All I do is try loading Smash Ultimate with ARCropolis installed.</br>

### Crash Report

[01599373016_01006a800016e000.log](https://github.com/Atmosphere-NX/Atmosphere/files/5179151/01599373016_01006a800016e000.log)</br>

### System Firmware Version

10.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: Official 5.3.2
- Do you have additional kips or sysmodules you're loading: None, it was a fresh install of Atmosphere
- Homebrew software installed: ARCropolis for Smash Mods

### Additional context?

- Additional info about your environment:
- After I get this error, sometimes actions relating to the SD card become unresponsive, such as launching the Homebrew channel and loading homebrew apps. Perhaps I should reformat my SD card? It's already FAT32 with 64KB clusters?

masagrator, 2020-09-06:
```
It's already FAT32 with 64KB clusters?
```
For best performance on Switch use always 32kB.

WuBoytH, 2020-09-06:
Well I guess I'm gonna reformat my sd card, then, I'll comment on this if it works

WuBoytH, 2020-09-06:
Update: I have reformatted my SD card to FAT32/32kB, and it works now. I guess it's worth mentioning that some guides for getting into Smash Ultimate modding recommend formatting to FAT32/64kB.

closed 2020-09-06 by WuBoytH