Atmosphere-NX/Atmosphere #1139: fatal error occured when running atmosphere (titleid 0100000000000005)

VashTS000, 2020-09-07:
[report_0000000000c65118.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5183861/report_0000000000c65118.zip)

abort() called (0xffe). just recently updated to 10.1.0 firmware. same files were used on another switch successfully with no issues. using atmosphere 0.14, both clean and with sigpatches. used hekate 5.3.2 and fusee-primary bootloaders, all fail with the same issue.

SciresM, 2020-09-07:
You improperly performed a partial update, 10.x package2 was not successfully flashed to mmc.

Fix is to manually write 10.x package2 to NAND, seek support in Nintendo Homebrew or ReSwitched discord for help with this.

closed 2020-09-07 by SciresM

VashTS000, 2020-09-07:
thank you. i suppose it's time to move to daybreak and make more frequent backups.