Atmosphere-NX/Atmosphere #1140: Nintendo Switch getting hot when connected to wi-fi on cfw

Chico3589, 2020-09-07:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ N/A ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ My nintendo switch gets hot everytime i connect to the internet, i used tesla menu to monitor my temperatures and cpu usage and cpu#3 was at 100% when connected to the internet, as soon as i turn airplane mode it goes back to 30ish%. ]</br>
[ At first i thougth i did, buut after a clean install of atmosphere the issues persisted]</br>
[ N/A ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ I just downloaded the latest version and it started happening, not sure if it can be reproduced but here is some information: i am running atmosphere on my nand with the latest switch version possible, i only have the homebrew appstorebesides the included atmosphere homebrews ]</br>

### Crash Report
N/A
[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

10.1.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?
N/A
- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?
N/A
- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Nanolx, 2020-09-12:
I can confirm the issue on my switch, it's also heating up.

It seems to also cause minor battery drain in standby mode it seems, it consumes about 4 - 5 % overnight, where on official firmware was about 1 - 2 % (nothing major, but maybe still noteworthy).

ELY3M, 2020-09-15:
I go online in atmosphere all of the times.  I use ZERO protections.  no 90dns used 
I can tell you that my switch do get hot.   I thought it is just old switch.

closed 2020-09-17 by Chico3589

yifanfu, 2021-01-23:
I'm having the same issue here with the latest 0.17.1. I also have 90dns and incognito enabled, not sure which one of them caused the issue :)