Atmosphere-NX/Atmosphere #1142: A fatal error occured when running atmosphere

malekalhalabi, 2020-09-14:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

GGLinnk, 2020-09-14:
I'm not sure but I think it can be an archive bit issue, if you use hekate there is an option to fix that.

Nthehunter, 2020-09-18:
Yeah , I have the same issue. I update atmosphere for the problem of freezing in mario 3d all stars, exactly when I out of mario 64 the console freezing. So, when I update this errors appear to me, I solved it, copy and paste the previous pack of atmosphere that I have.

I add the fatal error report:

[report_0000000006560c59.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5245551/report_0000000006560c59.zip)

SciresM, 2020-09-18:
Closing because the original issue creator didn't use the issue templates or upload a report. Reopen using them.

@Nthehunter re-extract latest atmosphere zip. You overwrote part of atmosphere with an outdated copy of mission control.

closed 2020-09-18 by SciresM