Atmosphere-NX/Atmosphere #1143: A fatal error occurred

juzzash, 2020-09-15:
[image]

Attached is the error I’m getting when trying to launch atmosphere. Please see below details... any help as to why it’s not launching at all?

Switch firmware 10.2.0
I’ve tried both atmosphere 0.14.3 & 0.14.2
128 GB SD, formatted to exFAT

I payload it using fusee-gelee.firebaseapp.com/# and the fusee-primary.bin file that comes with atmosphere.
The only nro I use is EdiZon and Checkpoint. #

SciresM, 2020-09-15:
Upload the report file as a zip, or to some other filehost then link it here.

juzzash, 2020-09-15:
Actually, worked it out thanks to a previous post! 
I connected the SD to my pc, highlighted all files and folders except Nintendo, and turned off the archive. It now boots properly!

closed 2020-09-15 by juzzash