Atmosphere-NX/Atmosphere #1145: Fatal error after updating to ofw 10.2.0 and updating atmosphere to 0.14.3

assessmentservices, 2020-09-15:
## Bug Report

### What's the issue you encountered?

I recently updated my switch firmware to 10.2.0 and then my Atmosphère to 0.14.3 and hekate to 5.3.2

Whenever I try to run syscfw from hekate or on its own I get a fatal error. This happens every time and I can not get around it.

It will say the following

A fatal error occurred when running Atmosphere.
Title ID: 010000000000002b
Error Desc: std : : abort ( ) called (0xffe)
Report saved to /atmosphere/fatal_errors/report_00000000092d12a8.bin
Press POWER to reboot.

### How can the issue be reproduced?
This issue occurred when I tried to update and it happens every time I attempt to run syscfw

### Crash Report

There don't appear to be any crash reports but I will include the bin file.
atmosphere crash report.zip (https://github.com/Atmosphere-NX/Atmosphere/files/5226582/atmosphere.crash.report.zip)

### System Firmware Version

10.2.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Both fusee and hekate
Fusee is the one included in the zip for the latest atmosphere update and hekate is 5.3.2 https://github.com/CTCaer/hekate/releases/tag/v5.3.2 but this issue was occurring on a previous hekate release too
- Official release or unofficial build: official
- Homebrew software installed: I only use JKSV

SciresM, 2020-09-15:
Same error as #1124, the mount name one for erpt that doesn't make any sense.

If you can, please send me your erpt-sysmodule save file. This is save file `0x80000000000000D1` in SYSTEM partition.

assessmentservices, 2020-09-15:
> 
> 
> Same error as #1124, the mount name one for erpt that doesn't make any sense.
> 
> If you can, please send me your erpt-sysmodule save file. This is save file `0x80000000000000D1` in SYSTEM partition.

I'm sorry I don't see that on my SD anywhere. Could you tell me how I can find it please?

SciresM, 2020-09-15:
That file isn't on your SD card. Like I said, it's in the SYSTEM partition. You'll need to mount that (with e.g. hekate UMS + HacDiskMount) to get it.

assessmentservices, 2020-09-15:
> That file isn't on your SD card. Like I said, it's in the SYSTEM partition. You'll need to mount that (with e.g. hekate UMS + HacDiskMount) to get it.

Thanks for clarifying. I got hacdiskmount and I'm trying to figure this out so I can get you what you need. I'll follow up when I do.

assessmentservices, 2020-09-15:
I'm really sorry, I couldn't figure out how to do what you wanted me to do. However I was able to work around this problem by installing a previous version of Atmosphere. I went back to 0.13.0 and was able to boot it up even while on OFW 10.2.0. Hopefully that'll help others with this problem until someone can get you the info you need.

GGLinnk, 2020-09-18:
I can try to reproduce the bug and upload the system 0x80000000000000D1 save if you need it @SciresM.

Jacquelinett, 2020-09-19:
I think I'm having the same issues. I tried to downgrade down to 0.13.0 but that didn't seem to fix it however.

Everything was working fine until I updated the oFW this morning. Updating to latest atmosphere didn't seem to work.

I use fusee

[report_0000000006376019.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5250115/report_0000000006376019.bin.zip)

SciresM, 2020-09-22:
@Jacquelinett you are not having the same issue, you need to run the archive bit fixer in hekate.

Klepuna420, 2020-09-22:
I have the same error for a while. On your SD card is a folder called 010000000000002b. Just delete this folder and the error should not appear anymore. Did work for me.

SciresM, 2020-09-22:
@Klepuna420 That deletes the atmosphere component that's crashing. I've told you to please upload your erpt system save so that I can debug/actually fix the problem...

Anyway, I deleted your suggestion, because it will interfere in people providing me with information I need to solve the problem.

Jacquelinett, 2020-09-22:
Thank you SciresM, seem like that was indeed the issues, which is strange because I was able to run Atmosphere prior to this and never had an issues up/had to fix bit until the latest system update. I'm not sure what happened for that to happen

But thank you again! It really helped

beefsquatchbeef, 2020-09-23:
im getting an idential error output as the original post in this thread.

here is my error report file:  

[report_0000000008920739.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5265212/report_0000000008920739.bin.zip)

SciresM, 2020-09-23:
@beefsquatchbeef So you are. I would really appreciate if you could dump your erpt savegame for me. Please use HacDiskMount (+ Hekate UMS to expose NAND over USB) to mount the SYSTEM partition, then send the `0x80000000000000D1` save file.

beefsquatchbeef, 2020-09-23:
i've tried to follow your instructions above, and but don't see a `0x80000000000000D1` save file listed anywhere

SciresM, 2020-09-23:
@beefsquatchbeef SYSTEM is encrypted, HacDiskMount has a UI for entering keys (dumpable with lockpick rcm). Maybe join the ReSwitched discord for help getting it mounted?

beefsquatchbeef, 2020-09-24:
thanks for the reco -- i believe i was able to do this successfully. here's the file:

[80000000000000d1.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5272860/80000000000000d1.zip)

SciresM, 2020-09-24:
Thank you very much @beefsquatchbeef, this is now fixed in atmosphere-master. 

Zip containing fixed erpt-sysmodule (extract to your SD card, replacing the 0.14.4 release one) 
[here](https://github.com/Atmosphere-NX/Atmosphere/files/5273156/atmosphere-0.14.4-erpt-fix.zip).

closed 2020-09-24 by SciresM

VGPete, 2020-10-17:
Thanks for the erpt fix SciresM. This is exactly what i needed and i was a mess trying all the different suggestions. I appreciate all you do.