Atmosphere-NX/Atmosphere #1147: "An error has occured" - 2162-0002

RenCE1, 2020-09-15:
## Bug Report

### What's the issue you encountered?

After uploading Atmosphere and Hekate to the latest versions and updating my Switch, I ended up getting this error.

### How can the issue be reproduced?

Once into Hekate, launch the system's CFW.

### Crash Report

https://drive.google.com/file/d/1f6SSx_ky3cABziMod2I5pa4bim6a0WMk/view?usp=sharing
(Here is the zip. I tried to start it up three times so I think they're all the same error.)

### System Firmware Version

10.2.0.

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: Official release- the atmosphere-0.14.3-master-28ceedb5+hbl-2.3.4+hbmenu-3.3.0.zip.
- Do you have additional kips or sysmodules you're loading: No.
- Homebrew software installed: [ PPSSSPP, Retroarch, hbmenu  ]

### Additional context?

- Additional info about your environment: N/A

SciresM, 2020-09-15:
File is inaccessible.

RenCE1, 2020-09-15:
https://drive.google.com/file/d/1f6SSx_ky3cABziMod2I5pa4bim6a0WMk/view?usp=sharing @SciresM

SciresM, 2020-09-15:
Delete your custom theme by removing the /atmosphere/contents/0100000000001000 folder.

closed 2020-09-15 by SciresM

dollerbill, 2021-06-04:
Thank you @SciresM, same issue on 12.0.2 and 0.19.3, but easily resolved