Atmosphere-NX/Atmosphere #1151: Daybreak hangs at 0%

Abbotta4, 2020-09-16:
## Bug Report

### What's the issue you encountered?

Daybreak hangs at 0% while trying to upgrade EmuMMC from 10.0.2 to 10.2.0

### How can the issue be reproduced?

I followed the steps [https://nh-server.github.io/switch-guide/extras/updating/](here) to dump firmware from my SysMMC and used that update data with Daybreak to try to update to 10.2.0. I get through all of the beginning dialogs, but Daybreak hangs for me at:
"Installing update
0% complete
Beginning update setup...
Update setup complete.
Preparing update..."

### Crash Report

N/A

### System Firmware Version

10.2.0 OFW
10.0.2 CFW

### Environment?

bootloader: Hekate 5.3.3 from release page

### Additional context?

N/A

SciresM, 2020-09-16:
I sincerely doubt it hangs, having used daybreak to update to 10.2.0 the other day.

However, in emummc the initial Preparing Update phase can take up to a minute or two before it actually begins installing. 

Be patient.

closed 2020-09-16 by SciresM

djshrimper, 2021-07-17:
Did you manage to get past this? It's happening to me aswell trying to update from 10.0.2

Awalsh053, 2024-01-25:
for me it sat a 0% for about half hour then jumped to 80% just be patient