Atmosphere-NX/Atmosphere #1154: 2 Game specific Crashes

HenryHeliolisk, 2020-09-17:
on the latest version of atmosphere: 0.14.3,
When you play super Mario all stars, and play Super Mario 64, if you close out using the select press menu, the entire console crashes.

Also, for super smash brothers ultimate, when It starts up, everything is fine but when I select my character to fight, the music still plays but the screen goes dark, I heard the this is a sysmodule problem, but I need to know how I would go about fixing it.

fincs, 2020-09-17:
Duplicate of https://github.com/Atmosphere-NX/Atmosphere/issues/1153

closed 2020-09-17 by fincs

cheesycod, 2020-09-17:
> Also, for super smash brothers ultimate, when It starts up, everything is fine but when I select my character to fight, the music still plays but the screen goes dark, I heard the this is a sysmodule problem, but I need to know how I would go about fixing it.

Try removing all custom system modules and try again. Next time, please use the ReSwitched Discord as this is not a support forum