Atmosphere-NX/Atmosphere #1155: fusee-primary boot slower than ever before!

azoreseuropa, 2020-09-17:
I noticed that fusee-primary is much slower! I decided to timer fusee-primary and Hekate - CFW (SYSNAND).. The big difference: fusee-primary boot 26 seconds and CFW (SYSNAND) through Hekate boot 16 seconds.

SciresM, 2020-09-17:
This isn't really a bug, or even an issue.

Pinging @hexkyz for thoughts, but yeah I would describe this is incredibly low priority.

closed 2020-09-17 by SciresM

hexkyz, 2020-09-17:
It's not fusee-primary, but rather fusee-secondary (specifically the nxboot code path). Unfortunately, as the project grows bigger boot times will get longer, that is, unless some optimizations are put in place. Hekate already optimizes the boot process with multiple techniques so I'm not sure if we want to spend time on this. Either way, this is definitely low priority for the time being.