Atmosphere-NX/Atmosphere #1156: (another) fatal erroe 10.2.0 / 0.14.3

morpheus2502, 2020-09-17:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
fatal error / no boot in cfw possible after updating to 0.14.3 and 10.2.0
0.14.3 worked fine with 10.1.0; also with the nes sigpatches
tried w/o exFat SD card as well

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?
boot into cfw -> sytsm stuck

booting to ofw works fine

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report
so such file or dir

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version
10.2.0

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?
fusee
included in the github 0.14.3 zip

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?
tittle id: 0100000000000008
Error Desc: std::abort() called (0xffe)

addes 4 bin files which should be the error reports

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

[Documents.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5239369/Documents.zip)

SciresM, 2020-09-17:
Partial update/you extracted older mission-control on top of latest atmosphere.

Re-extract the latest atmosphere zip in its entirety.

morpheus2502, 2020-09-17:
inalso did that...
i made a clear install without anything exept atmosphere

morpheus2502, 2020-09-17:
[report_0000000005e2f279.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5239507/report_0000000005e2f279.zip)

another try...ending up with another error ....

morpheus2502, 2020-09-17:
hi i did that...made a clean install of atmosphere...i posted another report in the closed issue.

the error occur before i tried several things (incl older mission) to downgrade to 10.1.0...
no success so far for me.

so the first time it occur i just did that procedure:
10.1.0/0.14.1
update to 10.1.0/0.14.3
recognized that tinfoil is not working
updated sigpatch, reboot
10.1.0/0.14.3 worked fine (also tinfoil)
update to 10.2.0/0.14.3
result: this error...

took a blank sd card/ fat32/ only arhomaphere 0.14.3 on it...also error.

hope you can help
me out;-)

fincs, 2020-09-17:
> recognized that tinfoil is not working
updated sigpatch, reboot

Piracy is not supported.

SciresM, 2020-09-17:
The crash reports you posted result from you not fully extracting ams. The boot2 binary generating the crash report is not the atmosphere 0.14.3 boot2.

Piracy is not supported.

If you encounter other issues after you actually update atmosphere properly, post the crash reports.

SciresM, 2020-09-17:
Oh, and the most recent report you posted is "Corrupt FAT filesystem for SD card".

Fix that.