Atmosphere-NX/Atmosphere #1157: Fatal error on 0.14.3 and 10.2.0

GrandEmpress66, 2020-09-17:
## Bug Report

I keep getting an error report when trying to boot up into Atmosphere CFW.
I have uploaded to firmware 10.2.0 on the Switch and manually updated CFW to 0.14.3
I've never had this isue before now and am fairly green in this area, so I'm not too sure what to do here.  Hopefully someone can help me

### What's the issue you encountered?

Title ID:  010041544d530000
A fatal error occurred when running Atmosphere.
Title ID:  010041544d530000
Error Desc: std : : abort ( ) called (0xffe)
Report saved to /atmosphere/fatal_errors/report_0000000005c3e89.bin
Press POWER to reboot.

Can't get past the boot screen and Atmosphere logo.  Never had this issue happen to me before.

### How can the issue be reproduced?

This happens every time I attempt to boot into CFW.  It stops right after the Atmosphere logo 

### Crash Report
I'm not really sure how to work this part

### System Firmware Version

10.2.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:  Fusee
- Official release or unofficial build: Official
  
- Do you have additional kips or sysmodules you're loading: Not that I'm aware?
- Homebrew software installed: JKSV and Checkpoint are all I really use

SciresM, 2020-09-17:
Please upload the error report file your issue mentions.

skywalker3li, 2020-09-17:
Hi, I also got the same error on 1.40.3 and tried 1.40.4 +10.2.0

error report is attached

[report_0000000005ec487a.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5242547/report_0000000005ec487a.zip)

SciresM, 2020-09-18:
@skywalker3li "same error" I really doubt it. You installed FAT32 firmware but your SD card is formatted as exFAT. Manually flash exFAT firmware to package2, or reformat your SD card as FAT32.

skywalker3li, 2020-09-18:
Shheeeesh yea it seems I screwed it up -__-

sorry but how do I "Manually flash exFAT firmware to package2"?

GrandEmpress66, 2020-09-18:
@SciresM I was going to ask the same thing.  My SD card is a 256 and neve gives me the option to go to FAT32, only exFAT.
How exactly would you " Manually flash the firmware?"  and what exacty IS package2?

SciresM, 2020-09-18:
@GrandEmpress66 you didn't upload your report. Please do so.

GrandEmpress66, 2020-09-18:
@SciresM 
Here's the report.  Hopefully I did this correctly

[report_0000000005c3e89e.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5242884/report_0000000005c3e89e.zip)

SciresM, 2020-09-18:
@GrandEmpress66 Fix your archive bits, hekate has a tool for this.

closed 2020-09-18 by SciresM