Atmosphere-NX/Atmosphere #1158: atmosphere 0.14.4 not working properly with 3d all stars; tinfoil shortcut isnt working also

EnriqueJuarez918, 2020-09-18:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ] Tinfoil and sig patches wont work with super mario 3d all stars but will boot with any other game and also applet 
[ Did you make any changes related to Atmosphère itself? ] updated the exefs & kip patches

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ] installed new atmosphere 0.14.4 with new patches, wont play tinfoil and 3d all stars
### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ] I think this is it but im not sure 100% https://gist.github.com/EnriqueJuarez918/3a4954948ded086183959a23663606eb

### System Firmware Version

10.2

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee
- Official release or unofficial build: 
  - [ atmosphere-0.14.4-master-fa0df994+hbl-2.3.4+hbmenu-3.3.0 ]
- Do you have additional kips or sysmodules you're loading: loader patches
- Homebrew software installed: [ pnes pplay and psnes and retroarch]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ] heres a build of my SD
[AMS-sigpatches-0.14.4-cfw-10.2.0.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5243257/AMS-sigpatches-0.14.4-cfw-10.2.0.zip)

Masamune3210, 2020-09-18:
Piracy is not supported

SciresM, 2020-09-18:
Can't reproduce, and piracy is not supported.

closed 2020-09-18 by SciresM