Atmosphere-NX/Atmosphere #1159: Error Code: 2168-0002 (0x4a8)

kleyox, 2020-09-19:
error shows this

[image]

masagrator, 2020-09-19:
LDN_MITM is crashing.

Read changelog from 0.14.2
https://github.com/Atmosphere-NX/Atmosphere/releases/tag/0.14.2

And now read readme of ldn_mitm

https://github.com/spacemeowx2/ldn_mitm

No support for Atmosphere 0.14.2+
Ask devs of ldn_mitm to update their stuff.

mysiki, 2020-09-20:
Same here, ok but what I suppose to do ? I don't use LDN_MITM and don't remember adding it (but probably yes....). 

It's build in Atmosphere, so A. V0.14.2 will be not fonctional. I thinks no.
Or do I need to remove it ? How ? (I know this is work from other team but here readme is really small :) 

Thanks for helping

masagrator, 2020-09-20:
> Same here, ok but what I suppose to do ? I don't use LDN_MITM and don't remember adding it (but probably yes....).
>
> It's build in Atmosphere, so A. V0.14.2 will be not fonctional. I thinks no.
> Or do [...]

If you have different "Program" ID than as on screen (4200000000000010), then you are not the same.

mysiki, 2020-09-20:
It was the same, and I just use the last version of LDN_MITM (than say compatible with 0.14.1 only) in same time. It work ! 

Sry for noise ... FIY, last version of LDN_MITM work (let start switch) with version 0.14.4.

brenodantas10, 2020-09-20:
Last ldn_mitm doesn't work after atmosphere 0.14.2. 

Delete it or you will get an infinite loading screen (or black screen) when trying to use anything related to the module (the ldnmitm_config or enabling wireless play at a game)

closed 2020-09-20 by SciresM