Atmosphere-NX/Atmosphere #1160: Game specific Bug: Moero Chronicles & Crystal menu lag.

Saborla, 2020-09-19:
## Bug Report

Moero Chronicles & Crystal menu lag.

### What's the issue you encountered?

Both games on the Moero series (Moero Chronicle Hyper / Moero Crystal H) lag dramatically on any in-game menus.
The games boot properly and run fine until you open any in-game menu, where they slow down considerably, more so the more populated the menu is. Once you close the menu, the game continues to run normally.
The problem persist even on a fresh SD card with only a clean copy of the latest atmosphere. Redownloading to system memory didn't help, either.

### How can the issue be reproduced?

Start Moero Crystal H, play until you can access any menu. Most noticeable by playing until you can access the item shop and trying to buy something.

### Crash Report

The game doesn't crash.

### System Firmware Version

Experienced on:
10.X
9.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: both fusee_primary or hekate_ctcaer_5.1.2 present the same issue.
- Official release or unofficial build: 
  - Initially experienced on Kosmos 15.1
  - Also tested on clean Atmosphere 0.14.4

SciresM, 2020-09-23:
This is solved now.

The game calls `nn::oe::GetDesiredLanguage()` for every text string, never cache-ing the result.

`nn::oe::GetDesiredLanguage()` calls `am::IApplicationFunctions::GetDesiredLanguage()`, which calls `nn::ns::GetApplicationSupportedLanguage()`, which creates a new session to `set` and calls `ISettingsServer::GetLanguageCode()`.

Atmosphere mitm's settings in order to allow for user-configurable languages, previously this was attempting to read an INI file off the SD card every call.

Games are *supposed* to cache the result of GetDesiredLanguage()...whoever programmed this game didn't get the memo, and calls the function for every text string, so all this crazy overhead was happening once per text string per frame.
I'm doing a two-fold fix.

1) I optimized this code to much more aggressively cache results, basically all the file overhead is solved this way. The only remaining overhead is the extra IPC layers associated with connecting to a mitm service instead of a non-mitm service.

I verified with 1) done, the game is significantly less laggy than before. However, it's still slightly slower than stock.

Thus, 2) I'm creating game patches to cache the result of nn::oe::GetDesiredLanguage(). These patches save ~5+ IPC calls to ~4 different system processes each invocation, which is done many times per frame, resulting in less lag/faster than even stock. Patch for Moero Chronicles Hyper is tested/verified working. Making the patch for Moero Crystal H now.

1) is committed in f4d10a4. I'll close the issue and attach the patches when done.

closed 2020-09-23 by SciresM

SciresM, 2020-09-23:
Patches [here](https://github.com/Atmosphere-NX/Atmosphere/files/5267843/atmosphere_moero_language_cache_patches.zip).

SciresM, 2020-09-23:
Note: with patches atmosphere gets noticeably better performance than stock in certain menus.

Compare results with those patches vs e.g. stock footage: https://youtu.be/wq03VU4gtfw?t=1478 [Embed]

Pay attention in particular to shop girl's tail, in stock the animation slows down due to all the overhead, with patches in ams it's totally smooth.

I'm going to consider this totally resolved now.

rsn8887, 2020-09-23:
What else is atm mitm’ing and can I turn that stuff off somehow?

Fzhek, 2020-10-24:
I had the same issue but with audio stuttering as well. The patches fixed the lag but the audio stuttering is there, especially in the voices. The audio is totally fine in the cut scenes. Any way to resolve this?

gavinyueng, 2020-12-02:
I use the Asian version（Moero H [0100D7E0110B2000][v0][BASE]）. I used your patch but it doesn’t work. What should I do?

Masamune3210, 2020-12-03:
Please don't ask for support if you are going to admit to piracy