Atmosphere-NX/Atmosphere #1161: Game specific Bug:title id 0100d9a005ed6000

twyl3513, 2020-09-20:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

10.2.0  0.14.4 atm
[crash_report.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5251093/crash_report.zip)

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Masamune3210, 2020-09-20:
fill out the issue report, it is there for a reason

twyl3513, 2020-09-20:
i attach the crash_report.zip in the issue,i installed the game's DLCs,then i start it.the game report error

Masamune3210, 2020-09-20:
But you didn't fill out the issue template, which has information that is required for a reason, like your Atmos version and switch firmware version

masagrator, 2020-09-20:
> But you didn't fill out the issue template, which has information that is required for a reason, like your Atmos version and switch firmware version

They provided fw and ams version, just not where it should be

SciresM, 2020-09-20:
Reopen using the issue templates.

closed 2020-09-20 by SciresM