Atmosphere-NX/Atmosphere #1162: Several Super Mario 3D All-Stars crashes and hangs

leo60228, 2020-09-20:
## Bug Report

### What's the issue you encountered?

1. Starting any game in 3D All-Stars has a chance of causing the game to freeze. When this happens, closing the game from the HOME Menu leads to "The software was closed because an error occurred." After this happens, trying again without a reboot leads to a fatal error. This seems to happen more often the longer the system has been running.
2. Opening the Homebrew menu from the album while 3D All-Stars is running leads to a panic.
3. Uploading a video to Twitter from the album while 3D All-Stars is running has a low chance of freezing the system at some point in the process, requiring a forced power-off.
4. Viewing Controls or Actions from the minus menu in 3D All-Stars has a moderate chance of freezing the system, requiring a forced power-off.

### How can the issue be reproduced?

1. Try starting a game in 3D All-Stars.
2. Start a game in 3D All-Stars, then go to the home menu and start the album with the configured HBL key combination.
3. Start a game in 3D All-Stars, then go to the home menu, open the album, and attempt to upload a video to Twitter.
4. Start a game in 3D All-Stars, press minus, then scroll down.

### Crash Report

None appear to have been produced for any of the issues.

### System Firmware Version

10.2.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Reproduced with both Fusèe and Hekate.
- Official release or unofficial build: Official release
  - 0.14.4
- Do you have additional kips or sysmodules you're loading: MissionControl
- Homebrew software installed:
 - Edizon 3.0.1 and 3.1.0 nightly
 - backtrace test homebrew
 - Cheats Updater
 - Daybreak
 - Fake News Injector
 - ftpd snap! 2.3.1-00c2c442
 - Goldleaf (only used for file management)
 - HB App Store
 - libnx irsensor and jit samples
 - JKSV
 - Kosmos Toolbox
 - Kosmos Updater
 - launchapplication test homebrew
 - Rust test homebrew
 - Lockpick
 - mtp-server-nx
 - nxdumptool 1.1.10 and 1.1.8 (I can't find 1.1.8 on my SD card...)
 - Reboot to Payload
 - two broken Rust test homebrews
 - sys-clk Editor
 - SysDVR Settings
 - VVVVVV-CE port
 - VVVVVV: Make and Play Edition port

### Additional context?

- emuMMC, sysMMC on 3.0.0
- Booted via PegaScape
- sysMMC has never been connected to the internet without a connection-blocking DNS

fincs, 2020-09-20:
Can you try a completely unmodified copy of Atmosphère, specifically without 3rd party system modules? I cannot reproduce the random freezes locally (except for the Album-HBL crash which I believe to be a known issue which affects Sunshine specifically)

leo60228, 2020-09-20:
I've wiped my Atmosphère installation and reinstalled it, I'll report back if I experience any more crashes.

leo60228, 2020-09-20:
Just had the first issue happen with a completely clean Atmosphère.

SciresM, 2020-09-20:
@leo60228 Not sure how to reproduce. Just opened/closed mario sunshine ~20 times in a row with no problems...

leo60228, 2020-09-20:
I can't easily reproduce this, but this happens multiple times a day for me.

SciresM, 2020-09-20:
"Just had the first issue happen with a completely clean Atmosphère."

"trying again without a reboot leads to a fatal error (0xf601)"

If you're getting a fatal error, you should have a fatal error bin, no?

leo60228, 2020-09-20:
I don't, oddly. Not sure why, but the `fatal_errors` folder is empty.

SciresM, 2020-09-20:
What file did hekate/fusee say it was saved to? Generate issue with both bootloaders to see if one saves file where the other doesn't?

Can't do anything without a report binary, which you indicate should be generated.

leo60228, 2020-09-20:
I'll report back if/when one of the issues happens again.

SciresM, 2020-09-24:
Closing, as none of these reproduce, and hbl-hang-when-sunshine-is-suspended was fixed in master.

closed 2020-09-24 by SciresM

OnkelM, 2020-09-28:
> Closing, as none of these reproduce, and hbl-hang-when-sunshine-is-suspended was fixed in master.

Does this also involve the slowines of hbmenu and all homebrew launched from?
Without mario sunshine loaded the bhl is working fine
If mario sunshine is loaded and i try to launch hbl it seems to hang for 1 minute until the apps are showing, but hbmenu is very laggy (scrolling and moving). and executing any homebrew that is also extremely slow. e.g. cheats updater will crash while extracting the cheats b/c of this lag

fincs, 2020-09-28:
> Does this also involve the slowines of hbmenu and all homebrew launched from?

The "hang" was just the slowiness. This is fixed in master.