Atmosphere-NX/Atmosphere #1164: Error 2001-0125 when launching hbmenu via app override on 4.0.1

mitchellmebane, 2020-09-22:
## Bug Report

With HOS 4.0.1 and Atmosphere 0.14.4, Atmosphere crashes when launching hbmenu via app override, similar to #905.

### What's the issue you encountered?

hbmenu crashes when trying to launch it via a game, though it works fine in applet mode.

### How can the issue be reproduced?

- Set up emuMMC running HOS 4.0.1 with Atmosphere 0.14.4
- Attempt to launch hbmenu by holding R while launching a game
- Wait on the Nintendo logo screen for about 40 seconds

### Crash Report

https://gist.github.com/mitchellmebane/21c02cf36dbcd5cb581d6d4ed97646b1

### System Firmware Version

4.0.1 on both sys and emu

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate 5.3.3 -> fusee-primary
- Official release or unofficial build: 
  - Official release version 0.14.4
- Do you have additional kips or sysmodules you're loading: None
- Homebrew software installed: N/A.  The crash happens even when the only things in my `/switch/` dir are `prod.keys` and `reboot_to_payload.nro`.

### Additional context?

The same issue occurs with switchbrew/nx-hbmenu@45efcfcb9869f0d1dae43c20672d420eec011181 compiled with libnx 3.3.0.

The crash dump of that, and the compiled hbmenu, are at https://gist.github.com/mitchellmebane/ce26f1d1c5806af0c735f93f9cee78e3

SciresM, 2020-09-22:
Not an atmosphere issue.

The problem is that am is trying to boost system memory limit (for automatic gameplay recording), but nx-hbloader has already allocated all of the application memory.

The root issue appears to be that nx-hbloader doesn't detect automatic recording on 4.x (the relevant block is inside hosversionAbove(5, 0, 0)): https://github.com/switchbrew/nx-hbloader/blob/master/source/main.c#L201

This should probably be changed in nx-hbloader.

SciresM, 2020-09-22:
Should be resolved by https://github.com/switchbrew/nx-hbloader/pull/32

closed 2020-09-22 by SciresM

mitchellmebane, 2020-09-23:
Indeed, I tried with a 1.0.0 game (Cave Story+) and the crash went away.  I compiled the new version of hbloader and switched back to my 4.0.1 game (Owlboy) and that worked as well.  Thanks!