Atmosphere-NX/Atmosphere #1165: Fatal error when running Atmosphere title ID 0100000000000001

slickrasta, 2020-09-22:
I keep getting the fatal error on title ID 0100000000000001 everytime I launch hbmenu (select Album) on the newest Atmosphere version. 

Currently on FW 9.0.0 and Atmosphere 10.0 and works fine with hekate 5.0.2 which works totally fine including hbmenu. All I'm trying to do is update to FW 10.2.0 via Daybreak so I can play my physical copy of All-Stars. Tried deleting atmosphere/sept/switch folders and putting fresh files. Even tried with a brand new SD card formatted to FAT32 with only the atmosphere files on it literally nothing else, exact same error every time I try to open hdmenu on new Atmosphere version using the fusee primary payload. Tried hekate archive bit fix already as well. I feel like I'm losing my mind I've tried so many things and am completely baffled how it can even happen on a brand new SD card with no other files whatsoever. Please help me.

Bin from current SD:
[report_00000000265510bc (current sd).zip](https://github.com/Atmosphere-NX/Atmosphere/files/5259519/report_00000000265510bc.current.sd.zip)
Bin from brand new SD: 
http://www.mediafire.com/file/hqwt5zdgvileuyg/report_00000000179ec795.rar/file

SciresM, 2020-09-22:
Please re-open using the issue templates.

closed 2020-09-22 by SciresM

RichardCranium0, 2026-09-30:
I'm having the same issue. What do you mean by "re-open using the issue template"?