Atmosphere-NX/Atmosphere #1166: Fatal error title ID 0100000000000001 when opening hbmenu on 0.14.4

slickrasta, 2020-09-22:
## Bug Report

### What's the issue you encountered?

I keep getting the fatal error on title ID 0100000000000001 everytime I launch hbmenu (select Album) on the newest Atmosphere version (0.14.4). Currently on FW 9.0 and Atmosphere 0.9.4 and works fine with payload Hekate 5.0.2 including hbmenu with no issues. All I'm trying to do is update to FW 10.2.0 via Daybreak so I can play my physical copy of All-Stars. 

I've tried deleting atmosphere/sept/switch folders and putting fresh files. Even tried with a brand new SD card formatted to FAT32 with only the atmosphere files on it literally nothing else, exact same error every time I try to open hdmenu on new Atmosphere version using the fusee primary payload. Tried Hekate archive bit fix already as well. I am completely baffled how it can even happen on a brand new SD card with no other files whatsoever.

### How can the issue be reproduced?

No idea if this is reproducible since I'm not seeing anyone else reporting this specific issue when I Google it / look here.
1. Format SD card to FAT32
2. Copy Atmosphere 0.14.4 files onto SD
3. Boot using fusee-primary.bin payload via TegraRCMGUI
4. Select Album icon, instantly crashes to this:
(https://user-images.githubusercontent.com/28207695/93851101-60d93c00-fc64-11ea-9f72-f3261731be82.jpeg)

### Crash Report

Bin from current SD (FAT32):
[report_00000000265510bc (current sd).zip](https://github.com/Atmosphere-NX/Atmosphere/files/5259519/report_00000000265510bc.current.sd.zip)
Bin from brand new SD (FAT32):
[report_00000000179ec795 (brand new sd).zip](https://github.com/Atmosphere-NX/Atmosphere/files/5259527/report_00000000179ec795.brand.new.sd.zip)

### System Firmware Version

9.0.0

### Environment?

- What bootloader: fusee-primary.bin
- Official release 0.14.4 (from this github)
- Do you have additional kips or sysmodules you're loading: No.
- Homebrew software installed: [ EdiZon, Daybreak, ChoiDujourNX]

SciresM, 2020-09-22:
@slickrasta Please contact me (SciresM#0524) on discord. 

I am highly confident this isn't an atmosphere issue, but an issue with your current firmware installation.

slickrasta, 2020-09-22:
> @slickrasta Please contact me (SciresM#0524) on discord.
> 
> I am highly confident this isn't an atmosphere issue, but an issue with your current firmware installation.

How do I message you on Discord? Just signed up and it asked me to create my own group or add an invite? Not seeing a discord link listed on this github from what I can find either. Also tried searching your name but I guess it's only letting me search in my group I just created....

slickrasta, 2020-09-22:
> @slickrasta Please contact me (SciresM#0524) on discord.
> 
> I am highly confident this isn't an atmosphere issue, but an issue with your current firmware installation.

First time Discord user, do I add you as a friend or is there some discord server I need to add? Tried looking on the main github page not seeing a discord link there.

SciresM, 2020-09-22:
Confirmed issue was partial update, NCM kip vs package1 had differing pre/post 9.0.0 which lead atmosphere to use incorrect service command ABI.

Issue was solved by having user do `enable_ncm = 1`, so that NCM was replaced by atmosphere's (which guaranteed same ABI for loader vs NCM), followed by an update to 10.2.0 using daybreak.

closed 2020-09-22 by SciresM

EfLemon, 2021-12-08:
@SciresM Podrias ayudarme de igual manera por favor, tengo el mismo problema y no logro resolverlo, intente buscarte en discordia pero no doy con tu perfil, muchas gracias, saludos.

Masamune3210, 2021-12-08:
Try following the same steps, might be the same issue

Masamune3210, 2021-12-08:
If that doesn't work make a new issue and fill out the template