Atmosphere-NX/Atmosphere #1168: Latest Atmosphere creates Smash Ultimate error

FroyF2701, 2020-09-23:
## Bug Report

### What's the issue you encountered?

After updating to the latest Atmosphere version (0.14.4), the latest firmware (10.2.0) and updating sigpatches; Super Smash Brothers Ultimate started to present an error. The error in question occurs after the Character Select Screen. After pressing "FIGHT!!", the screen goes black while the music keeps playing. The only way to get out of this error is pressing the HOME button and closing the software. The issue seems to only happen to this specific game.

### How can the issue be reproduced?

By running Super Smash Bros. Ultimate on the current Atmosphere version

### Crash Report

N/A.

### System Firmware Version

10.2.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe-primary
- Official release or unofficial build: 0.14.4
- Do you have additional kips or sysmodules you're loading: No (To my knowledge)
- Homebrew software installed: Homebrew App Store, Atmosphere, Tinfoil, Goldleaf, Reboot to payload, Checkpoint, Edizon

### Additional context?

- Additional info about your environment: 

The following observations come from attempts at fixing the issue:

* uninstalled game, dlc, updates then reinstalling did not work
* save data was not the problem
* replay videos was not the problem
* testing the start of a battle a couple of times causes corrupted data (same black screen and the music plays until you go back to home to close the program type of error)
* micro sd checking for error did not fix it
* deleting overlays folder did not fix it
* deleting emuiibo information (also another thing that interacted with smash bros in my opinion)

SciresM, 2020-09-23:
Upload a screenshot of /atmosphere/contents folder, please.

Jevooz, 2020-09-23:
[image]

This also happened to me, I reached out to Froy on reddit, here is a screenshot of my atmosphere/contents.

SciresM, 2020-09-23:
That is a *lot* of custom system modules. I count 7 of them.

Does issue occur without all the custom sysmodules?

Jevooz, 2020-09-24:
[image]

I am trying a fresh install of atmosphere and everything else (picture above).
My previous attempts have been on firmware 10.2.0, atmosphere 0.14.4, micro sd on exFat (yes I know, I'm sorry!)

I have to redownload smash to try it out.

Every couple of times that you go into trying to start a battle, when the screen goes black but the music still plays and you have to go to HOME to close the program, it created corrupted data so I had to make sure to delete everything and try to redownload.

Downloading the base and update, then checking for corrupted data showed to be clean.
Then I tried to do a battle and got black screened.
Checked data and my base and update found corruption.

This happened a couple of times over the course of the day.
Also probably some DLC might have their own issues of corruption or bad downloads that caused NCA problems but I tried it specifically with the base and update by themselves to prove this.

Will update in about an hour when the latest redownload finished on the fresh atmosphere install.

SciresM, 2020-09-24:
Just to clarify, since that was a lot of text: the errors described were with the first configuration you posted? You have not tried stock ams yet, and are working on that right now?

Jevooz, 2020-09-24:
Yes sorry about that, the errors described in the previous block of text I made were with the first config (first picture with lots of things on the content folder) and I am trying how things turn out with a fresh new atmosphere 0.14.4 (second picture).

I have to wait for smash to redownload so maybe half an hour.

Jevooz, 2020-09-24:
Well that took forever to reinstall. Also this post might be a long one.

I deleted everything from my micro sd card except for the nintendo folder.
I drag and dropped the newest atmosphere 0.14.4, sigpatches and some other usual CFW programs.
made sure that I had nothing smash bros related on the sd card through tinfoil and edizon (but made sure to keep the sigpatches/tickets intact)

redownloaded smash base and update (some dlc had some NCA issues that caused errors when checking software and made it find corrupted data)

went in with no prior save data, started a stock battle and it worked! I got into a battle

The only things on the micro sd card are:
- Atmosphere 0.14.4
- firmware 10.2.0
- checkpoint
- edizon
- goldleaf
- jksv
- mtp-server-nx
- choidujoir
- tinfoil

Then I installed the DLC, still got tons of "Failed to register placeholder NCA" errors (mostly on the costumes) but everything got installed except for some bad mirrors for a handful of spirit events on tinfoil JITS.

to continue the experiment I restored my previous save file with checkpoint, went into the game and it still worked!

The things that changed from this morning when I got the error and now are:

- no residual files from previous drag and drops of way past atmospheres
- no emuiibo files
- no tesla or overlay files
- no residual files from other programs (tinfoil, hekate)
- deleted some nca folders made by the download error
- saw way less folders on the atmosphere main and contents folders (left previous, right new)

I'm really not sure what caused the error when it happened to me yesterday but a fresh atmosphere install seemed to fix that. I will try to put things back together one by one to see if tesla, txt cheat files recreate the error in some way.

It definitely happened on my previous attempt to jump to 0.14.4 and 10.2.0, I remember playing smash on the weekend before doing this to continue playing mario (64 made it glitch and not start up properly so I decided to upgrade).

Let me know what you think. images below are the references from the previous bullet points.

[image]
nca folders made from error

[image]
root comparison

[image]
atmosphere comparison

[image]
contents comparison

Jevooz

SciresM, 2020-09-24:
Piracy is not supported.

Sounds like issue was resolved via clean atmosphere, though -- if you find the addition of some sysmodule/addon breaks stuff, probably report it to the developers of that sysmodule/addon ^^

closed 2020-09-24 by SciresM