Atmosphere-NX/Atmosphere #1172: Atmosphere crashes on bootup after updating to 10.1.0

GSBro, 2020-09-26:
## Bug Report

A fatal error ocurred when running atmosphere.
itittle ID: 010041544d530000
error desc: std::abort() called (0xffe)
Report saved to /atmosphere/fatal_errors/report_00000000559655.bin

### What's the issue you encountered?

After updated my firmware to 10.1.0 Atmosphere won't sucessfully launch anymore. I tried running the Archive Bit fixer via Hekate and still nothing.

### How can the issue be reproduced?

Launching Atmosphere 

### Crash Report

[report_0000000005e8c420.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5285916/report_0000000005e8c420.zip)

### System Firmware Version

10.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: 5.3.3
- Do you have additional kips or sysmodules you're loading: No ]

### Additional context?

HookedBehemoth, 2020-09-26:
You installed FAT32 firmware but your SD card is formatted as exFAT. Manually flash exFAT firmware to package2, or reformat your SD card as FAT32.

GSBro, 2020-09-26:
> 
> 
> You installed FAT32 firmware but your SD card is formatted as exFAT. Manually flash exFAT firmware to package2, or reformat your SD card as FAT32.

Thanks for your help, I'll give it a try. I'm not sure if I know how to manually flash firmware so is there any disadvantage to reformatting my SD card as FAT32?

Masamune3210, 2020-09-26:
Exceedingly greater chance of not having your sd card corrupt randomly thanks to N's exfat driver being horrible

GSBro, 2020-09-26:
> 
> 
> Exceedingly greater chance of not having your sd card corrupt randomly thanks to N's exfat driver being horrible
So I noticed online it says if your SD card is larger than 64gb I should choose exFAT (my SD Card is 128gb) but to fix my problem I need to format to FAT32 so I'm not sure if that would cause any issues.

Masamune3210, 2020-09-26:
My card is 256GB and is formatted as FAT32, works fine. Just backup everything that is on the card to somewhere else, format the card using hekate, and then copy the files back onto the sd card and you should be fine

GSBro, 2020-09-26:
Perfect, thank you for all the help! I can't believe I goofed and accidentally installed the firmware for the wrong format >_<

closed 2020-09-26 by GSBro