Atmosphere-NX/Atmosphere #1173: Atmosphere refuses to boot [Title ID 010000000000002B]

EmreTech, 2020-09-26:
### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
What I was doing is after I booted up with Hekake, I tried to launch Atmosphere after doing a full NAND Backup, and it had a panic attack and refused to boot.

[ Did you make any changes related to Atmosphère itself? ]</br>
I did not make any changes to Atmosphere.

### How can the issue be reproduced?

I'm not sure how to reproduce this problem.

### Crash Report

My crash report is in the fatal_errors folder and cannot be read.

### System Firmware Version

10.2.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
I used Hekate.
- Official release or unofficial build: 
Official Build version 5.3.3
  
- Do you have additional kips or sysmodules you're loading:
Nothing.
- Homebrew software installed:
I don't have any Homebrew software installed.

### Additional context?

I'm using a Nintendo Switch from July 2017, and my serial number is XAW1001944

SciresM, 2020-09-26:
"cannot be read"

upload the binary. I can parse it.

EmreTech, 2020-09-26:
Okay. Here's the file: 

[report_07931687.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5286856/report_07931687.bin.zip)

EmreTech, 2020-09-26:
I found the issue. I forgot to turn off Share Error Information in Settings. I never knew it would crash if it was on.

closed 2020-09-26 by EmreTech

SciresM, 2020-09-27:
This was actually the same issue as #1124, #1145, and was fixed recently. The next Atmosphere release won't trigger the problem.