Atmosphere-NX/Atmosphere #1175: Switch emummc restart to rcm when trying to go to sleep mode

MatteKalle, 2020-09-27:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
When pressing power button one time the switch instantly goes to rcm mode instead off sleep mode. Also when I start up I have to spam the home button because else it tries to go to sleep mode and then it goes to rcm mode.

[ Describe the issue in detail and what you were doing beforehand. ]</br>
Setup before begining:
Unpatched switch.
exfat formated sd-card
Firmware 10.1.0
Atmosphere 0.13.0
The error message about game cart slot appearing now and then.
Auto RCM
Probably some burnt fuses

So this is how it went down. I hope I remember to write all the crutial steps and the order which i did them in.

I located the necessary rawnand.bin, boot0 and boot1 for making the nand restore. This is a backup I made before going CFW. I dont know what firmware version I was on when I made this backup but I can see that the files are dated december 2017.

I downloaded hekate_ctcaer_5.3.2_Nyx_0.9.3 and put the bootloader folder on the SD card root.

I transferred a copy off all my SD files to my computer.

I started hekate by using hekate_ctcaer_5.3.2.bin.

I used "partition SD Card" in Hekate and created a HOS (FAT32) and emuMMC (RAW).

I created emuMMC SD Partion in Hekate and it says Enabled!.

Copied back all the backuped SD card files to the root off the SD card.

Tried to launch by entering RCM-mode and injecting fusee-primary.bin. It worked and I checked system firmware and saw that I was on E (emunand).

Tried going in to hekate again to launch from there but at first it did not work. Tried a bit and I think what got it working was when I added fusee-primary.bin to the payloads folder in bootloader folder (it was empty before) and also adding code to the hekate.ipl.ini. The code I added was some lines found in a forum thread. One for starting CFW (EMUNAND) and one for Stock (SYSNAND).

Renamed the hekate_ctcaer_5.3.2.bit to reboot_payload.bin and overwrote the file in the atmosphere folder (for being able to reboot to hekate).

Copied the backuped Nintendo-folder to emuMMC\RAW1 and comfirmed that my installed games was working again on emunand.

Now when emunand was working I was convinced it was safe to restore my nand backup. I followed this guide https://suchmememanyskill.github.io/guides/nandrestore/ and did not get into any trouble.

Entered RCM-mode and injected (the same) fusee-primary.bin as before sysnand restore but now nothing happens. The screen is black and TegraRcmGui shows that it is ready to inject again after like 2 seconds.

Trying to inject hekate_ctcaer_5.3.2.bin again and it works.

Trying to launch CFW (emunand) through Hekate exactly as I did before restoring the sysnand. Now it seems to start up, it shows the Atmoshpere logo, it shows the Nintendo Switch logo, it shows the battery status symbol and then for a short moment the sleep screen (the screen where you have to press a button 3 times for it to wake up). Here I have found that I have to spam home button or else it will go to rcm mode. When doing so everything seems to be working until I try to go into sleep mode again. Then it goes to rcm.

[ Did you make any changes related to Atmosphère itself? ]</br>
Not before just after to try to fix the problem. I have tried removing atmosphere and replace it by the newest version.

[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?
See the description above. Not sure though. I can't seem to find anyone else that had this problem.

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>
See above.

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
Can't find that folder. I suppose that means that it have not created any report.

[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

10.2.0 AMS 0.14.3 E
When setting up emunand I was on 10.1.0. Now I have updated to 10.2.0 but that did not change anything.</br>

[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
hekate ctcaer 5.3.3. From there I am running fusee-secondary (from the atmosphere folder).
I have also tried launching fusee-primary but I can't get that to work since after the sysnand restore.

- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
No
- Homebrew software installed: [ * ]
Tinfoil

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]
A separate (?) problem is that sysnand won't work either. When trying to start I get "Unknown pkg1 version. Hos version not supported! Failed to launch HOS". I have tried booting this with a new sd card and only the required stuff but I get the same message.
I have also tried redoing the restore off boot0 boot1 and rawnand but that did not change anything.
Also tried formating the main partition back to exfat but no success (now Im back at fat32 again).

SciresM, 2020-09-27:
I cannot reproduce on my end -- sleep works fine in my emummc partition and my emummc file-based.

`A separate (?) problem is that sysnand won't work either. When trying to start I get "Unknown pkg1 version. Hos version not supported! Failed to launch HOS". I have tried booting this with a new sd card and only the required stuff but I get the same message.

Sounds like sysmmc is possibly corrupted, though (or you're using outdated hekate)? 

When booting sysnand with fusee-primary, does sleep work?

MatteKalle, 2020-09-28:
Thank you for taking the time for this.

Do you have any suggestions what to try to get the sleep working on emunand again? If no suggestions I guess I try recreating emunand on a different sd card.

Also I tried starting sysnand with fusee-primary from hekate now with this ini-code.
emummc_force_disable=1
fss0=atmosphere/fusee-primary.bin
icon=bootloader/res/icon_switch.bmp
stock=1

Then I get this error message.
"
No mandatory secmon or warmboot provided!
Failed to launch HOS!
"

Suggestions? Should I try to repair boot0 in some way?

shchmue, 2020-09-28:
fss0= must point to fusee secondary not primary

MatteKalle, 2020-09-28:
Ok, that I have tried before and then I get the same error message as I already stated.
"
Unknown pkg1 version.
HOS version not supported!
Trying to backup bootloader...
Found pkg1 <´´>.
Unknown pkg1 version.
HOS version not supported!
Failed to launch HOS!
"

SciresM, 2020-09-28:
I was asking you to boot fusee primary, not fusee via hekate. Configure console to use sysmmc (you can use hekate for this), then boot via fusee-primary as payload without using hekate at all.

MatteKalle, 2020-09-28:
Ok, I will try that when I get home.

MatteKalle, 2020-09-28:
I have now disabled emummc from hekate and when trying to boot fusee-primary.bin (the latest one) I get the following message.
"
[ERROR] Fatal error: [NXBOOT] Couldn't parse boot0: Illegal byte sequence!
Press POWER to reboot
"

MatteKalle, 2020-09-30:
I've search for the above error message but I would really appreciate some guidance.
Instead I tried something more simple.

I tried backing up my emummc with Hekate (boot-files + rawnand).
Then I used another SD card and this time made the partitions with tegraexplorer.bin.
After that I placed the latest atmo and hekate and started with the latest hekate bin-file.
In Hekate I Created the emuMMC as SD Raw Partition.
I then restored the boot-files + rawnand to the SD-card.
When I now try to boot up cfw emummc I get hekate boot logo and then it goes directly in to RCM again.
After that I copied everything (except emummc folder) from my other SD card.
Tried again but the same thing happens when trying to go to emuMMC.
When I put in my other SD card again it is working as before = emuMMC boots but crash at sleep.

Any thoughts about this and/or the Fatal error -message is welcome.

MatteKalle, 2020-10-03:
Would really appreciate some guidance. I suppose next step would be a repair but I don’t know where to start.

MatteKalle, 2020-10-05:
Finally got sysnand working by getting live support from a kind person on Discord.
My backup off boot0 was faulty so that was the problem. When we looked at the file in a hexeditor it only showed zeros.
He helped me updating my firmware and to repair and restore my boot0.
Sysnand with online working as it should now =).

Next step will be to get emunand working. Will see if I can achieve that without needing to reinstall my games.
I'm thinking that I will start by doing a backup off my working boot0, boot1 and rawnand.bin.
Then I will try to restore only boot0 and boot1 to my emunand to see if that fixes the sleep error in emunand.
If not I suppose I will have to backup my game saves and create a completely new emunand from my working sysnand.
Please advice if you have any advice regarding this.

MatteKalle, 2020-10-06:
So to close this thread up I can tell you that emunand is working again.
I did not have to do anything more to get it working. Obviously the sysnand fix also fixed the emunand sleep problem.

closed 2020-10-07 by SciresM

pabz88, 2020-11-03:
Hi, was wondering if you could help me with the same issue. I am trying to dual boot the sxos and atmosphere via hekate. I downloaded all the latest files and updated the firmaware to 10.2.0 of both sys and emuNAND. I followed a guide in Gbatemp and have done the steps indicated there. Booted up to hekate and booted ams from there. The screen loaded and it said nxboot couldn't parse boot0. I'm trying to find a discord server to help me with this but as of now I don't have any luck.

Sleino85, 2020-12-11:
I have the same issue with this fatal error. Can one you help me out here?