Atmosphere-NX/Atmosphere #1177: Fatal error occurred when running atmosphere (title id 0100000000000005)

Achintha1929, 2020-09-30:
## Bug Report

### What's the issue you encountered?

After updating my Switches firmware using ChoiDujournx I get a blue screen when booting normally.
The only way to boot into ofw is rcm and I cannot boot into Atmosphere at all without getting this error:</br>
a fatal error occurred when running atmosphere
title id: 0100000000000005
error desc: std::abort() called (0xffe)
report saved to /atmosphere/fatal_errors/report_000000000777cacc.bin]</br>

### How can the issue be reproduced?

This error occurred after the firmware was successfully installed and I hit reboot.
Before the switch wouldn't even rcm into Hekate but I was able to fix that by
reformatting my sd card from exFAT to fat32, restoring my NAND backup, reinstalling
Atmosphere and updating Hekate from 5.0.1 to 5.3.3</br>

### Crash Report

https://drive.google.com/file/d/1vwcSku1dIioaQUidUuMC2ump3xE1vgTn/view?usp=sharing</br>

### System Firmware Version

9.2.0 -> 10.2.0 (The settings in ofw says that the current version is 10.2.0)</br>

### Environment?

- What bootloader (fusèe, Hekate, etc) was Atmosphère launched by: Hekate 5.3.3 bin
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]: Official 0.14.4
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: [Edizon and Checkpoint]

### Additional context?

I also tried trying to re-upgrade the firmware using ChoiDujour.exe but the script always fails in finding the keys.

SciresM, 2020-10-01:
@Achintha1929 You improperly performed a partial update, 10.x package2 was not successfully flashed to mmc.

Fix is to manually write 10.x package2 to NAND, seek support in Nintendo Homebrew or ReSwitched discord for help with this.

closed 2020-10-01 by SciresM

Achintha1929, 2020-10-02:
Thank you soooooo much @SciresM your solution worked and I was able to rebuild the MMC using https://suchmememanyskill.github.io/guides/unbrick/ Cheers!!!