Atmosphere-NX/Atmosphere #1178: "The software was closed because an error occurred!" Yu-Gi-Oh! Legacy of the Duelist Link Evolution keeps crashing.

NaturalHarmonia, 2020-09-30:
## Bug Report

### What's the issue you encountered?

[ I was in a duel and about to use a card when the game froze and then I got 'The software was closed because an error occurred'. This has also happened with Tinfoil and it happens frequently. This didn't happen a few days ago. ]</br>
[ N/A ]</br>
[ N/A ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ I don't know exactly. It seems to occur from just naturally playing the game.]</br>

### Crash Report

https://gist.github.com/NaturalHarmonia/22d53af4b2eb91e606cf40718b122d40

### System Firmware Version

10.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee
- Official release or unofficial build: Official release
  - [ Official release version Atmosphère 0.14.4 ]
- Do you have additional kips or sysmodules you're loading: No.
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ N/A ]

Masamune3210, 2020-09-30:
Piracy is not supported

nwhiteman42, 2020-09-30:
And you know they didn't buy the game because?

HookedBehemoth, 2020-09-30:
> [ I was in a duel and about to use a card when the game froze and then I got 'The software was closed because an error occurred'. This has also happened with Tinfoil and it happens frequently. This didn't happen a few days ago. ]
Check the edit log

closed 2020-10-01 by NaturalHarmonia