Atmosphere-NX/Atmosphere #1179: Remove wifi IC check.

musiccankill, 2020-09-30:
### What feature are you suggesting?
#### Overview:
-Bypass check of wifi IC on boot</br>

#### Smaller Details: 
- If the wifi IC (hardware component) has failed, there is no way to boot into horizon so maybe you can disable the check and allow it to boot even if the component is faulty/missing.</br>

### What component do you feel this would best fit within?
- [Exosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Fully-featured custom secure monitor.</br>
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

### Why would this feature be useful?
A working console even without working wifi is better than a brick console, so users with faulty wifi ICs will benefit from that.</br>
Consoles with this problem need a wifi IC from a nintendo switch donor board as far as i know as commercial replacements will not work (possibly custom firmware in the wifi IC?).

SciresM, 2020-09-30:
This sounds like it would involve heavy editing to wlan sysmodule.

I don't know how you would do this, but I would happily accept a pull request to add the patches for this feature.

closed 2020-09-30 by SciresM

Hitori638, 2026-05-17:
This would be really helpful in my case. My console wont boot into OFW or CFW because of a partially failing chip. Bluetooth still works but WiFi does not. I am happy to help with testing or provide any info or logs if that would be useful.