Atmosphere-NX/Atmosphere #1180: A fatal error occurred when running Atmosphere Title ID 010000000000002b

Leonelvader, 2020-10-01:
[image]
#528 # What's the issue you encountered?
Title id: 010000000000002b 
### How can the issue be reproduced?
When I try to inject the Payload fusee primary with Emunand activated it loads the Atmosphere screen and then I get that error
A fatal error occurred when running atmosphere
Title ID: 0100000000000002b 
Error Desc: std: abort () (called 0xffe)

### Crash Report
https://gist.github.com/Leonelvader/75ec5b22e94d5ba150958ea0c0a4be06
### System Firmware Version

10.2.0</br>

### Environment?
Fusee 0.14.0
Atmosphere 0.14.4 Official release

nwhiteman42, 2020-10-01:
Don't use video guides.  They become out of date very quickly

SciresM, 2020-10-01:
@nwhiteman42 Please don't try to provide support for issues you don't understand.

@Leonelvader This is actually the same issue as #1124, #1145, #1173 and was fixed recently. The next Atmosphere release won't trigger the problem.

Zip containing fixed erpt-sysmodule (extract to your SD card, replacing the 0.14.4 release one) 
[here](https://github.com/Atmosphere-NX/Atmosphere/files/5273156/atmosphere-0.14.4-erpt-fix.zip).

closed 2020-10-01 by SciresM

infiniteichiwon, 2020-10-01:
@SciresM I got a fatal error saying: 
Failed to stat stage2 (atmosphere/fusee-secondary.bin)!

SciresM, 2020-10-01:
@infiniteichiwon Extract latest atmosphere release zip to your SD card.

infiniteichiwon, 2020-10-01:
Ok thank you I'll try that out

infiniteichiwon, 2020-10-01:
> @infiniteichiwon Extract latest atmosphere release zip to your SD card.

it doesnt give me the error anymore thankfully but I cant pull up the mod menu for some reason.

infiniteichiwon, 2020-10-01:
@SciresM sorry if your busy. When I load the game the mod menu doesn't, appear any reason why ?

SciresM, 2020-10-01:
I have no idea what you mean by "mod menu".

To override a game with the homebrew menu, hold R when launching it.

shoaley, 2020-10-07:
> @nwhiteman42 Please don't try to provide support for issues you don't understand.
>
> @Leonelvader This is actually the same issue as #1124, #1145, #1173 and was fixed recently. The next Atmosphere [...]

I'm still getting the same error

HookedBehemoth, 2020-10-07:
That is not the same issue as the one you opened prior.
It's the same title (erpt) that fails though.

Relastra, 2021-01-28:
indeed I copy pasted new erpt-sysmodule abd still I get the same issue. Any news here ?

SciresM, 2021-01-28:
@Relastra the sysmodule linked in this issue is deeply outdated. Upgrade back to latest atmosphere release, then post your fatal error binary.

deckyfx, 2021-05-02:
i also get this error any help?

YondaimeKazzy, 2022-02-06:
@SciresM Hello, excuse me, but this is definitely not fixed. At least not for me and apparently not for deckyfx either. I did every step in here. I reinstalled Atmosphere (and hekate!), didn't work. I tried the erpt-sysmodule, didn't work. I deleted the erpt-sysmodule and reinstalled Atmosphere, didn't work. I'm getting a bit upset now, I've been at this for two days. I followed this guide to change sd cards, because I got an sd card with more space, but it's still not working. I don't know what I've done wrong, but it would be very helpful if someone could help me. 

Guide I followed: https://nh-server.github.io/switch-guide/extras/transfer_sd/

Skonikol, 2022-02-06:
> @SciresM Hello, excuse me, but this is definitely not fixed. At least not for me and apparently not for deckyfx either. I did every step in here. I reinstalled Atmosphere (and hekate!), didn't work. I tried the erpt-sysmodule, didn't work. I deleted the erpt-sysmodule and reinstalled Atmosphere, didn't work. I'm getting a bit upset now, I've been at this for two days. I followed this guide to change sd cards, because I got an sd card with more space, but it's still not working. I don't know what I've done wrong, but it would be very helpful if someone could help me.
> 
> Guide I followed: https://nh-server.github.io/switch-guide/extras/transfer_sd/
> 
> EDIT: Also something to note, when I boot up my switch the normal way (like not cfw), it seems to say my microsd card isn't working. I have a Samsung EVO Plus card, which is weird because I just checked on google and it says it's a good sd card for the nintendo switch

You should check the card for errors and operability. For example AxoFlashTest, F3X, F3 and the like.

YondaimeKazzy, 2022-02-07:
> You should check the card for errors and operability. For example AxoFlashTest, F3X, F3 and the like.

Did all of that, the card was fine, so I decided to try changing the sd cards once more. using the same guide. I followed everything and did exactly what it said, but it still happened. I don't understand where it's going wrong, exactly.

Masamune3210, 2022-04-02:
SD reader on switch could be going bad, does hekate say anything about 1 bit mode or anything about SD errors in general?