Atmosphere-NX/Atmosphere #1183: Atmoshpere Crashes when selecting Data Management "manage software" and "move data between" menus

Tac24, 2020-10-09:
## Bug Report

error code:2162-0002 (0x4a2)
Program:0100000000000039
Firmware:10.0.4 (Atmos at 0.14.4)

### What's the issue you encountered?

Atmoshpere seems to crash when accessing the native Data management menus "manage software" and especially "move data between SD card and storage (a feture introduced naively to Horizon OS on 10.0 which can be handy even for emunand)

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

I can get it to work proper if I run the one game on the emunand that's in the SD card spot, but if I select "Move data between" it crashes at the "checking for free space" portion...

and afterwards selecting manage software and move data will crash- especially when running a homebrew NSP (mario64) or homebrew menu. also seems to crash easier if I run a game from the emunand or game cart instead

### Crash Report
https://i1.lensdump.com/i/0RZ1Dz.jpg

### System Firmware Version

Atmos:0.14.4
Switch:10.0.4

### Environment?

used to use Hekate, but started using tinfoil and of course Tinfoil doesn't play nice with Hekates booting, so switched to Fusee_Primary to boot atmos

[image]
. I hate how strict tinfoil is but its pretty seemless and great for updates when it works. Goldleaf and Awoo can't compare

going back to Hekate doesn't help this sadly.

### Additional context?

something unusual I've noticed, the program number is 0100000000000039.... something I can't seem to find in latest atmos updates. a quick and sloppy search tells me it has something to do with fonts of custom themes... something I've never ran before.

my SD card reader also has had some issues in the past, I've had to replace the part recently. Yet I took it apart and fixed the module and it reads fine now on the switch and on computers.

fincs, 2020-10-09:
Piracy is **not** supported. Homebrew is exclusively supported as NRO files running under nx-hbloader.

closed 2020-10-09 by fincs

GGLinnk, 2020-10-11:
It does the same with homebrew like Goldleaf (Installed NSP Version), same with any kind of homebrew forwarders, impossible to move them from one memory to another...