Atmosphere-NX/Atmosphere #1187: Fatal Error Occurred When Running Atmosphere

beaniemachinie, 2020-10-13:
## Bug Report
I've been working on homebrewing my switch all day, but I cannot get Atmosphere to load no matter what I do. I've tried deleting and re-extracting and everything under the sun, but no matter what I do it just does not work. 

### What's the issue you encountered?

Whether I inject the hekate payload or fusee-gelee payload, I cannot load in Atmosphere. It brings up all the logos like it's going to, but then it goes to a black screen that says: 
A fatal error occurred when running Atmosphere.
Title ID: 010000000000002b
Error Desc: std: :abort() called (0xffe)
Report saved to /atmosphere/fatal_errors/report_00000000080b5963.bin

### How can the issue be reproduced?

Inject Hekate/ fusee-gelee payload through RCM 
Attempt to launch through hekate CFW

### Crash Report

I don't have the crash report (because I deleted my files to go back on my switch) but if ou need one I can do it again 
### System Firmware Version

Firmare is version 10.2.0

### Environment?

I used the most recent versions of both hekate and Atmosphere straight from github, along with TegraRCMgui 

### Additional context?

I've already reformatted my SD card to FAT32 from exFAT, but that did not solve the issue of Atmosphere not loading.

HookedBehemoth, 2020-10-13:
The crash report is very much needed to diagnose this issue.

beaniemachinie, 2020-10-13:
Cool beans. I’ll go ahead recreate it and post it

carcaschoi, 2020-10-13:
https://github.com/Atmosphere-NX/Atmosphere/files/5273156/atmosphere-0.14.4-erpt-fix.zip

unzip to your SD card

beaniemachinie, 2020-10-13:
Well, I went ahead and reloaded everything to try and recreate my issue but now it's loading fine?? I'm gonna leave this thread open just in case something else happens

SciresM, 2020-10-13:
I'm gonna close this because there's no unresolved issue.

closed 2020-10-13 by SciresM