Atmosphere-NX/Atmosphere #1188: [ERROR] Fatal Error: Failed to intialize the SD card!.

UltraAlpha, 2020-10-15:
## Bug Report

### What's the issue you encountered?

Atmosphere NX failed to boot up after injecting the payload into my Switch's SD card. I was using an Onn 256GB exFAT card, and my Switch was connected to my laptop using a common USB-C to USB-A cable.

### How can the issue be reproduced?

I injected the fusee-primary.bin file into my Switch through the TegraRCM GUI, and I got this error.

My SD Card is an Onn 256GB with exFAT. I am not using Hekate, and the Atmosphere bundle I am using is the most recent version of Atmosphere NX. I am doing a fresh install of the Atosphere CFW onto my Switch, and no other hacks have been installed onto my system prior to this.

### Crash Report

None. My fatal error folder is empty. The error message I recieved when the console booted said:

[ERROR] Fatal Error: Failed to intialize the SD card!.

Press POWER to reboot.

### System Firmware Version

10.2.0

### Environment?

Bootloader: fusee
Atmosphere NX Official Release version 0.14.4
TegraRcmGui_v2.6_portable
Homebrew software installed: None

### Additional context?

I was following instructions from this video, on an un-modified, vanilla Nintendo Switch. My Switch has been determined to not be patched. 

https://www.youtube.com/watch?v=qCiYSqjYOqU

masagrator, 2020-10-15:
First don't use exfat. Use fat32.

Is your sdcard detected when booting without cfw?

UltraAlpha, 2020-10-15:
I am preparing to reformat my card to fat32, but since it's a 256-gig card, it would take several hours to create a backup of its contents and then reinstall it.

As for running the card without cfw, there aren't any issues. As I said before, my switch wasn't jailbroken in any way before I installed Atmosphere. This is my first dive into the Switch modding scheme.

SciresM, 2020-10-15:
Does hekate successfully access the card?

UltraAlpha, 2020-10-15:
> Does hekate successfully access the card? 

I... Don't have Hekate set up yet. 😅

SciresM, 2020-10-15:
Test it?

UltraAlpha, 2020-10-16:
I'm going to try installing and setting up Hekate right now. Should I still reformat my SD Card to FAT32, or is EXFAT fine?

Dark-Mind, 2020-10-16:
> 
> 
> I'm going to try installing and setting up Hekate right now. Should I still reformat my SD Card to FAT32, or is EXFAT fine?
> 
> EDIT: So after I drag and drop the bootloader into my SD Card's root, do I use the Hekate BIN file in place of the Fusee BIN file when I open up TegraGUI?

Yes, use hekate bin instead of fusee

Dark-Mind, 2020-10-16:
Anyhow I've had this error before. And it seems to be that HOS system (your main OS) is fat32 only.

You need to download a system update. And choose exfat support when installing.

Or just format to fat32 like what the devs said.

UltraAlpha, 2020-10-16:
> Anyhow I've had this error before. And it seems to be that HOS system (your main OS) is fat32 only.
> 
> You need to download a system update. And choose exfat support when installing.
> 
> Or just format to fat32 like what the devs said.

Thank you, I'll do that once I finish formatting my card to FAT32. I'll comment if anything changes.

MYSTlKO, 2020-10-16:
> Thank you, I'll do that once I finish formatting my card to FAT32. I'll comment if anything changes.

Any results? Because I have the same problem.
SD card works fine with OFW, but get issues with CFW.  With fusee I can get into CFW, but after several minutes of playing it crashes with SD card errors. Hekate even won't let me load into atmosphere, because it can't recognize the SD card. 
The issues started suddenly. Haven't done anything before.

rapichekkansiwish, 2020-10-17:
The same problem. When trying to launch Atmosphere 14.4 from Hekate 5.3.3. the above error appears. I tried running fusee-primary payload without hekate, the result is the same. There are no logs. microsd is Kingston Canvas Go! Plus 256gb, FAT32. OFW 10.2.0. Updated OFW to support exfat, before flash CFW. On the 8GB card everything works fine, and when I boot into stock in hekate too.
Sorry for my bad English

SciresM, 2020-10-17:
...so hekate is working and able to access the SD card? That's what I was asking, not if fusee-from-hekate worked.

rapichekkansiwish, 2020-10-17:
Yes, Hecate is working and can access the SD card. After I click on boot Atmosphere, I get the error "[ERROR] Fatal Error: Failed to intialize the SD card!".

SciresM, 2020-10-17:
@rapichekkansiwish yeah, I think the hekate sd card driver has slightly better support than the fusee one.

Boot using hekate in your case, I guess, or get a new SD card.

rapichekkansiwish, 2020-10-17:
@SciresM Okay, thanks for the help

palmerasta, 2020-10-20:
I'm having trouble with the latest atmosphere build in emummc.
the system freezes randomly.
I do not have mod of any kind it is a clean system ...
sd 128gb kington in fat32 ...

UltraAlpha, 2020-10-20:
So, I tried to load my backup of my Switch data after I formatted the SD card. The strain was too much for my PC, and the boot files were corrupted. I had to go to the Geek Squad to get it straightened out.

So yeah... I'm not gonna be trying anything like that anytime soon. Consider this "issue" resolved.

MYSTlKO, 2020-10-20:
Like I said I had have the same issue. 
Figured out, that my SD card reader was broken. Two pins were missing so the SD couldn't work properly with the CFW. 
Have to fix it.

closed 2020-10-20 by SciresM