Atmosphere-NX/Atmosphere #1191: Atmosphere wont start up after updating to 10.2.0 Exfat

disturbia96, 2020-10-19:
### What's the issue you encountered?

 i recently updated my switch to 10.2.0 eXfat and i have A.M.S 14.4. but now if i start up the AMS i get this error code:
``
Title ID: 010041544d53000
Error [dest:](url) std::abort()called (0xffe)
report saved to atmosphere/fatal_errors/report_0000000005f4384d.bin

when i updated my switch with exfat32 my installed games didnt work, so i tried to install it with exfat and after that i keep getting this error.

does any one know how to fix this?
im desperate.. thanks!

### How can the issue be reproduced?

inserting my sd card and start up atmosphere

### Crash Report

report_0000000005f4384d.bin

### System Firmware Version

10.2.0
AMS 14.4

SciresM, 2020-10-19:
Please upload your error report binary.

disturbia96, 2020-10-19:
[report_0000000005f4384d.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5404463/report_0000000005f4384d.zip)

vaccarieli, 2020-10-19:
I had a similar issue what i did to fix it, was using hekate, and going to > tools > fix archive bit.

SciresM, 2020-10-19:
You installed non-exfat firmware but have an exfat sd card. Format your sd card as fat32, or manually flash exfat firm to package2.

closed 2020-10-19 by SciresM

disturbia96, 2020-10-19:
I tries to format my SD card to fat32 but I'm unable to do so.  Only Exfat of nnd something, how do I manually flash my exfat firm to package 2?

HookedBehemoth, 2020-10-19:
@vaccarieli You shouldn't try to give support for issues you don't understand.

Masamune3210, 2020-10-19:
@disturbia96 you can format your card using hekate, or most formatting tools on pc other than the built in ones in windows

disturbia96, 2020-10-20:
I got it working now but now when i want to install a game i get an error