Atmosphere-NX/Atmosphere #1193: [Pokemon Sw/Sh] Mods are crashing the game on a normal Atmosphere boot.

GamingInfinite, 2020-10-21:
## Bug Report

### What's the issue you encountered?

The issue I am having is that mods seem to not work with Atmosphere (when normally booted).  What I mean by normally booted is that I loaded rcm, sent the hekate payload, hit launch, hit sysnand cfw because I don't have a large enough sd card for anything else.  When you boot into atmosphere in that way it seems to break everything for some reason.  Although booting fusee-primary works just fine.  Before this started occuring, I was just playing normal pokemon shield.  I'm not sure what could be causing it since game patching is meant to be a built in atmospher feature.

### How can the issue be reproduced?

Like how I stated above.  Just boot atmosphere "normally" and then try and run pokemon shield with mods.  Simple as that.

### Crash Report

N/A.  Atmosphere itself didn't crash.

### System Firmware Version

Latest Switch Firmware.  Atmosphere is the latest one that's on sdsetup.  hekate is 5.3.3 which I think is latest.</br>

### Environment?

I launched a payload named fusee primary from hekate, not sure if that makes a difference, or even what the difference is.  Noticed the note here which makes me think that it is a bootloader based thing.

### Additional context?

I have Isle of Armor installed, although my mods all work on fusee-primary hekate launch so I don't think that matters.

Masamune3210, 2020-10-21:
What is the exact error that it gives you

GamingInfinite, 2020-10-21:
It just says unexpected error has occured.  Please return to the menu.  Then proceeds to show the corrupted data prompt, but then I do the scan and nothing's corrupted.  I just restart the console go back to fusee-primary payload off of hekate.

SciresM, 2020-10-21:
1) Upload the crash report generated on game open if you want support.

2) Atmosphere only officially supports fusee primary; you should seek help with the hekate developers if it works on fusee primary but not hekate.

closed 2020-10-21 by SciresM

GamingInfinite, 2020-10-21:
Literally the only way to open fusee primary (this was even on the official nx switch hacking website) is through hekate.  I don't understand your second point.  And the bug report instructions said to only put a crash report in the event that atmosphere itself fails altogether.  Also it's not a hekate problem, it's an atmosphere problem.  I'm smart enough to know that much.  How I understand it hekate is a bootloader, and doesn't do anything besides launching YOU'RE cfw os.  I can look and see if there is a crash report though.

SciresM, 2020-10-21:
"only way to open fusee primary is through hekate"

False, you send fusee primary as RCM payload instead of hekate. Fusee-primary can be chainloaded via hekate, but it is not designed to be loaded via hekate.

Atmosphere does not develop hekate. It does not officially support bugs that only reproduce when launching via hekate; it only officially supports fusee-primary.

Any issue that occurs when atmosphere is launched via hekate but not fusee-primary is a hekate configuration bug, not an atmosphere bug.

Please do not presume to correct me about how my project works.

And, further, if I ask you to upload a crash report, upload the crash report. Do not tell me you do not have to, when I am explicitly instructing you to do so.