Atmosphere-NX/Atmosphere #1194: [ERROR]Fatal error: Failed to initialize the sd card with micro sd 512Gb

EGLimon, 2020-10-22:
## Bug Report

N/A

### What's the issue you encountered?

I get the error : "[ERROR]Fatal error: Failed to initialize the sd card" with a micro sd card: HP mx330 UHS-I U3 512GB and when I press POWER to reboot the message changes to "[ERROR]Fatal error: Failed to initialize the file system: I/O error". 
And if I tried the same process with another micro sd ADATA  UHS-I class 10 A1 256GB it works fine.

### How can the issue be reproduced?

-Downloaded the latest version of atmosphere 0.14.4 
-Copied everything to sd card root
-injected payload fusee-primary.bin

### Crash Report

N/A

### System Firmware Version

10.2.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: tried with fusee and hekate 5.3.3 and get same error.
- Official release or unofficial build: 
       Atmosphere 0.14.4 official release
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: N/A

### Additional context?

- Additional info about your environment:
  - The micro sd is detected in official firmware
  - I also tried earlier versions of atmosphere from 0.12.0 to the latest injecting the fusee primary related to the version, but I had the same issue

closed 2020-10-22 by EGLimon

EGLimon, 2020-10-22:
I reformated the micro sd and selected specifically the cluster at 256kb and execute hekate 5.3.3 then CFW(sysMMC) on Launch section

fahmisora, 2021-08-08:
> I reformated the micro sd and selected specifically the cluster at 256kb and execute hekate 5.3.3 then CFW(sysMMC) on Launch section

Hi sir. Can you show me a quick video on how you solve this? I'm having the same problem using my 512GB sd card. 
Or like a list of steps on how you solve this. I'm still quite new