Atmosphere-NX/Atmosphere #1195: A fatal error occurred when running Atmosphere

Daniel17173, 2020-10-23:
A fatal error occurred when running Atmosphere

Title ID: 0100000000000008
Error Desc: std abort()called(0xffe)
Report saved to /atmosphere/fatal_errors/report_000000005fb4dbb.bin
press POWER to reboot

ns version 10.2

hekate_ctcaer_5.3.3.bin

tf atmosphere 0.14.4

Daniel17173, 2020-10-23:
[report_0000000005fb4dbb.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5429916/report_0000000005fb4dbb.zip)

Daniel17173, 2020-10-23:
Problem solved, just need to change the file in the contens/0100000000000008

HookedBehemoth, 2020-10-23:
You really shouldn't do this.
The 0100000000000008 folder contains atmosphere's boot2 reimplementation.
Certain CFW extensions and custom sysmodules won't work if you remove it.

SciresM, 2020-10-23:
@Daniel17173 your issue was probably caused by using an older version of mission control.

Extracting latest atmosphere release + updating mission control would fix it, and that's what you should do.

closed 2020-10-23 by SciresM