Atmosphere-NX/Atmosphere #1197: Multi-mod selection

MarioPossamato, 2020-10-30:
## Feature Request

### What feature are you suggesting?
#### Overview:
Allow the user to store multiple mods in their `sd:/Atmosphere/contents/<titleid>` folder.
The way it would work, is basically the same as SDCafiine, but with different folder names:
```
sd:
└───Atmosphere
    └───contents
        └───<titleid>
            ├───
            ├───
            ├───
            └───
```
```
                  .: Atmosphere :.

Select your option, then press the A key to launch.

-> sd:  Mod 1
   sd:  Mod 2
   sd:  Mod 3
   sd:  Mod 4
   sd:  Mod 5
   sd:  Mod 6
   sd:  Mod 7
   sd:  Mod 8

Page 1/2, press L/R to change.
```

#### Smaller Details: 
- [ These may include specific methods of implementation etc. ]</br>

#### Nature of Request:  
[ Remove all that do not apply to your request. ]
- Addition
  - [ Ex: Addition of certain original features or features from other community projects. ]
  - [ If you are suggesting porting features or including features from other projects, include what license they are distributed under and what, if any libraries those project use. ]
- Change
- Removal
  - [Ex: Removal of certain features or implementation due to a specific issue/bug or because of low quality code, etc.]

### What component do you feel this would best fit within?
- [Fusée](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's custom bootloader.</br>
- [Exosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Fully-featured custom secure monitor.</br>
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>
- [**Thermosphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's emuNAND implementation.</br>
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

[ Note: **Bolded components are not implemented** or are still at the prototyping phase. ]

### Why would this feature be useful?
[ If this is a feature for an end-user, how does it benefit the end-user? ]</br>
[ If this feature is for developers, what does it add to Atmosphère that did not already exist? ]</br>

masagrator, 2020-10-30:
PS: there is homebrew for it

https://github.com/nadrino/SimpleModManager

SciresM, 2020-10-30:
Yeah, this is something a user homebrew can do without needing atmosphere support.

closed 2020-10-30 by SciresM