Atmosphere-NX/Atmosphere #1200: Support for switch with > 4Gb of RAM

pontostroy, 2020-10-31:
I have switch with replaced ram chips(two 3 gb instead of two 2 gb ). It works fine but I can use only 4 GB.
I tried to rebuild atmosphere from source with this patch https://pastebin.com/LDaY3ek0 but look like a console turn of after atmosphere logo.

masagrator, 2020-10-31:
Do you even know how C++ works? Your changes broke code. Did you not get any warnings?

scrubslayer0104, 2020-10-31:
How the hell did you increase your ram?

CompSciOrBust, 2020-10-31:
> How the hell did you increase your ram? 

My best guess is he reballed the chips unless he somehow got a hat unit but I think they have 8 GB. If this is anything like when people used x360 dev kits to play retail games, games will just ignore the extra memory. Maybe homebrew can take advantage of it but I don't know anyone else who has extra ram so all homebrew is written to work fine with 4 GB.
Might be nice for running extra sys-modules or expanding the applet pool but I doubt 99.9% of users will do this mod given this is the first guy to ask AMS to support it. Personally I'd like to see Atmosphere detect the amount of ram and automatically support it without needing a special case for each configuration but it may not be worth Scires' time.

SciresM, 2020-10-31:
Atmosphere literally already supports higher ram based on value-reported-by-memory-controller.

So does stock HOS.

Your hang is because the memory controller does not consider the extra RAM accessible, and so you cannot read/write to it.

closed 2020-10-31 by SciresM

pontostroy, 2020-11-09:
It does boot with some patches and fss0, and can use all 6 GB.

SciresM, 2020-11-09:
"ok"