Atmosphere-NX/Atmosphere #1202: std::abort() called (0xffe) atmosphere

chinojed, 2020-11-02:
I just updated my atmosphere to 0.15.0. It was working fine for a while then encountered this error.

I've also tried reformatting the SD card to ExFat to no avail.

I appreciate any advise on this. Thank you!

[report_00000000058e4e68.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5474990/report_00000000058e4e68.zip)

SciresM, 2020-11-03:
@chinojed 1) you did not use the issue templates.

2) your issue comes from failing to actually update atmosphere.

You may have improperly used an older version of mission control which distributed a custom, out of date "boot2" system module, which overwrote the one from Atmosphere 0.15.0.

Re-extract latest atmosphere release zip.

closed 2020-11-03 by SciresM

chinojed, 2020-11-03:
Thanks @SciresM ! Managed to get it to work after what you suggested, and by fixing the archive bits set as well. Cheers!

souhaiebtar, 2021-06-13:
@SciresM thank you, using your suggestion, i downloaded control menu archive (to check the file that are used) and removed them manually, now every thing seem to be working thanks