Atmosphere-NX/Atmosphere #1203: Increase slots for title override

cucholix, 2020-11-03:
Currently there’s 8 slots to assign title ID and button to override game with hbmenu [0 to 7], would it possible increase these slots? 

It would be useful if you are using modified hbmenu that allows boot .nro homebrews directly from HOS menu, like a forwarder.

HookedBehemoth, 2020-11-03:
Maybe HBL override could be moved to the program specific ``config.ini`` instead since it's already parsed on every launch.

SciresM, 2020-11-03:
What use case requires more than eight configs?

For memory related reasons the limit must be a fixed number and not dynamic.

fincs, 2020-11-03:
Also fwiw hbmenu is not meant to be bypassed. "Forwarders" are explicitly not a supported use case in nx-hbloader.

closed 2020-11-11 by SciresM