Atmosphere-NX/Atmosphere #1206: [Mesosphère Bug] Sunshine Dock Issues

piepie62, 2020-11-12:
## Bug Report

### What's the issue you encountered?
When either inserting or removing the Switch into/from the dock with Sunshine running, Sunshine freezes.

### How can the issue be reproduced?
See above

### Crash Report
N / A

### System Firmware Version
10.2.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: either fusèe or hekate (both tested)
- Official release or unofficial build: Official M.15.0
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Not relevant

### Additional context?
N / A

masagrator, 2020-11-12:
Have you tried to check if this happens without mesosphere?

piepie62, 2020-11-12:
Yes: it does not. Scires asked me to open an issue, as he is currently unable to test anything related to a dock and I could find no other way to reproduce this

TheGinGear, 2020-11-13:
Is the game installed to the SD or the eMMC? and what format SD? I can attempt to reproduce

piepie62, 2020-11-13:
Yep, those are the exact symptoms. I didn't think to mention it, but I'm using a cartridge, so that's all three locations that it occurs from

doug0162, 2020-11-20:
I have also reproduced the issue, but only on the south half of Delfino Plaza where you spawn after booting the game. The bug doesn't occur if the switch is docked or undocked while Mario is on the north side of the plaza (where there is sunlight at the beginning of the game), and the camera is also facing north. I also docked and undocked in both Bianco Hills and Ricco Harbor areas, and didn't see the bug occur at either of those locations at all.

SciresM, 2020-12-08:
Issue confirmed not to reproduce in 0.16.0.

closed 2020-12-08 by SciresM