Atmosphere-NX/Atmosphere #1207: Cheat Writer Issue

Vegeta780, 2020-11-17:
Hi, Sorry am new and I don't know how to make a issue 
my issue is simple when
1- i made a cheat file more then a 100kb or 68kb my cheat codes did not load into Tesla or Edizon Se 
if i make a small file Then its load and work fine 
2-  i make a small file and My cheats code looks like  this 
[inf HP (total lines is = 75) ]
08100000 35035380 74370101 1D6A3CD2
its a cheat writer who write values first 64 lines into address but after 64 line writer  did not execute 65 line and never write a line  
is this problem with atmosphere cheat writer ?
sorry for bad English 
Current Version  10.2.0|AMS 0.15.0|S (hateke using )

SciresM, 2020-11-17:
The Cheat virtual machine has an instruction memory limit of 0x1000 bytes.

Each individual cheat has an instruction memory limit of 0x400 bytes.

For lines like the one you posted, which are four u32s, each line is 0x10 bytes, so it is is correct that you can only have 0x40 = 64 lines.

This isn't really a problem...cheats just aren't supposed to be that big.

Have you considered e.g. making a game patch, or optimizing your assembly to use a loop or something?

closed 2020-11-17 by SciresM

Vegeta780, 2020-11-18:
Thanks a lot for  response  my codes are not for looping but  is there a instruction game patch for my addresses are different values  because my codes are 13493 lines so kindly can you guide me how to game patch thanks

masagrator, 2020-11-18:
> Thanks a lot for response my codes are not for looping but is there a instruction game patch for my addresses are different values because my codes are 13493 lines so kindly can you guide me how to game patch thanks

It seems that you are writing directly to heaps. Making game patches means patching binary which is completely different approach.
If you have list of pointers instead of heap offsets you can make a tesla trainer using dmnt:cht to enable/disable various cheats. You need to know C++ though.

Vegeta780, 2020-11-18:
thanks but is there any tesla trainer example ? so i can make game patch