Atmosphere-NX/Atmosphere #1214: Fatal error 0x615 (2021-0003) on 0100000000001000

shayaantx, 2020-11-27:
## Bug Report
n/a

### What's the issue you encountered?

Upgraded to 0.15 and 10.2 and updated sig patches and i keep getting crash due to 0100000000001000 even though I deleted it from /atmosphere/titles and /atmosphere/content 

The crash occurs before switch 

### How can the issue be reproduced?

upgrade to 0.15 and 10.2

### Crash Report

[01606461966_0100000000001000.log](https://github.com/Atmosphere-NX/Atmosphere/files/5606512/01606461966_0100000000001000.log)

### System Firmware Version

0.15.0
10.2.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate 5.1.3
- Official release or unofficial build: official (from this git)
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: let me know if you really need this

bootloader = hekate 5.1.3

### Additional context?

I updated with 0.14.4/10.2 sig patches prior and it worked fine, then I used 0.15.0/10.2 sig patches and it stopped working. Prior to both sig patch upgrades, I upgraded atmosphere from 0.14.0 to 0.15.0, got a crash on same program id, deleted 0100000000001000 in /atmosphere/titles and /atmosphere/content, then everything worked fine.
Then I upgraded to 10.2 switch firmware and many games stopped working, so I updated sig patches once to 0.14.4/10.2, and one game worked, rest didn't. Then 0.15.0/10.2 and now i keep getting the above crash.

SciresM, 2020-11-27:
"No custom sysmodules." I don't believe you. The error in question is "service ran out of sessions", only happens with sysmodules taking up service slots. Delete atmosphere folder and re-extract latest zip.

Also, piracy is not supported.

closed 2020-11-27 by SciresM

shayaantx, 2020-11-27:
that worked, thanks