Atmosphere-NX/Atmosphere #1217: 11.0.0 Support

SciresM, 2020-12-01:
This issue is so that I can keep track of what I need to do to support 11.0.0, in no particular order.

- [ ] Diff secmon
- [ ] Make exosphere changes
- [ ] Diff kernelldr and kernel
- [ ] Make mesosphere changes (should this be temporarily delayed for 0.16.1 or something? TODO: evaluate)
- [ ] Diff boot sysmodule, update ams.boot for changes
- [ ] Implement sm::UserInterface command 4
- [ ] Update erpt::IContext command 10, new command 11
- [x] Update result handling for new erpt context ID shit
- [ ] Update pgl for new command 21
- [ ] Check boot2 for anything new/important we need (capmtp? we may be okay because re-used program id)
- [ ] Check fatal for anything new/important we need
- [ ] Check ro for anything new/important we need
- [ ] Check creport for anything new/important we need
- [ ] Update fusee-secondary to recognize/boot 11.0.0
- [ ] Update emummc to support FS-11 for exFAT + non-exFAT
- [ ] Update anti-downgrade tables inside Loader.
- [ ] General system stability improvements to enhance the user's experience.

Zyico, 2020-12-01:
ya it wont work should have waited until 11.0.0 just get a error trying to run Atmosphere so hopeful can get a 11.0.0 update soon

sapycola, 2020-12-01:
Thank you so much @SciresM

jeffemail, 2020-12-01:
If im running version11 there is no way to install atmosphere?

If no can i somehow go back to another version so i can ?

SciresM, 2020-12-01:
@jeffemail wait for atmosphere to support 11.0.0, or restore a backup you made previously.

Zyico, 2020-12-01:
> @jeffemail wait for atmosphere to support 11.0.0, or restore a backup you made previously.

ty that help a lot

sapycola, 2020-12-01:
after support 11.0.0 it's possibile to "offer a coffe"?

DrewV123, 2020-12-01:
I'm new here, so I was wondering how long does it normally take you to release updates?

masagrator, 2020-12-01:
> I'm new here, so I was wondering how long does it normally take you to release updates?

June 15th

DrewV123, 2020-12-01:
> > I'm new here, so I was wondering how long does it normally take you to release updates?
> 
> June 15th

Yeeeeah..I don't believe that XD

Kai-Beast223, 2020-12-01:
June 15th is the release date of every atmosphere update the person above is indeed correct.

DrewV123, 2020-12-01:
So I have to wait until June to play my nsp again, okay, sorry for doubting you...

Skonikol, 2020-12-01:
> > > I'm new here, so I was wondering how long does it normally take you to release updates?
> > 
> > 
> > June 15th
> 
> Yeeeeah..I don't believe that XD

June 15th = ASAP

Kai-Beast223, 2020-12-01:
Drew if you need atmosphere to play your NSP you are a pirate and that's not supported stop being a brat. If you want to play it just buy it 👍

col82293, 2020-12-01:
I applied the update by mistake! right at the start of my weekend too :,( keep up the good work! cant wait to get back to my playthrough of Xenogears on retroarch..

maishiben, 2020-12-01:
is it true that the soonest the updated atmosphere comes out is going to be june 15th?

SciresM, 2020-12-02:
Merging this because we're done other than the dram id.

NOTE: Atmosphere doesn't work on 11.x yet, hbl/hbmenu need updates.

closed 2020-12-02 by SciresM