Atmosphere-NX/Atmosphere #1218: 11.0.0 erpt changes

SciresM, 2020-12-01:
Right now erpt is updated for the new ABIs, but the new context tracking logic is not yet implemented.

The new logic for tracking resource limit limit + peaks is also not yet done.

We should implement it, so that we're more accurate.

ThePsycrow, 2020-12-01:
This issue (by itself at least) wouldn't cause the Switch to get "dirty" in a way that there's a ban risk when switching off of Atmosphere back to the default operating system, would it?

Just8n12, 2020-12-01:
Can I restore if my backup file missing few rawn file?

Just8n12, 2020-12-02:
:( my switch is currently banned from nintendo when i clicked on update software. Now my frimeware is 11.0 and my switch is banned. What a nice life :(

Masamune3210, 2020-12-02:
One, these comments don't go here, please stop. 
Two, you always risk getting banned when using cfw, the expectation should be getting banned, not the abnormality

cwsb69, 2020-12-02:
All i want for Christmas is i can play my switch nsp again, on firmware 11.0.0 :(

commit 6da28f4a (2020-12-03): erpt: update for 11.0.0 (closes #1218)
closed 2020-12-03 by SciresM