Atmosphere-NX/Atmosphere #1219: Fatal Error: [NXBOOT] Unable to identify package1!

Santiav90, 2020-12-01:
## Bug Report

### What's the issue you encountered?

So I tried to inject tha fusee-primary payload like always but now its giving me the Fatal Error: [NXBOOT] Unable to identify package1!

I tried to erase everything and reinstall atmosphere with the latest version and it still gives me the same error

### How can the issue be reproduced?

In my case, try to upload a payload

### Crash Report

It hasn't generated any reports, it doesn't even initialize atmosphere

### System Firmware Version

X.X.X</br>

It is updated to the latest version as of 1/12/2020 (december 1st)ºº

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee-primary
- Official release or unofficial build: official version 15 (latest)
- Do you have additional kips or sysmodules you're loading: no, just loading atmosphere and nothing else
- Homebrew software installed: nothing else

### Additional context?

Nothing more to add, as I am trying to install atmosphere from 0 but it is not working

SciresM, 2020-12-01:
11.0.0 released yesterday and isn't supported yet.

Wait for support, or restore a backup you made prior to updating.

closed 2020-12-01 by SciresM

Santiav90, 2020-12-01:
why did my issue get closed?

Santiav90, 2020-12-01:
ok thanks

Zeyzers, 2020-12-02:
> 11.0.0 released yesterday and isn't supported yet.
> 
> Wait for support, or restore a backup you made prior to updating.

How long does it usually take?

SciresM, 2020-12-02:
approximately negative five minutes from when you posted that.