Atmosphere-NX/Atmosphere #1221: Replace soft shutdown process

Nazosan, 2020-12-02:
## Feature Request

### What feature are you suggesting?
#### Overview:
Atmosphere replacing the stock shutdown process with one that is better.

#### Smaller Details: 
It should include a screen up until everything is properly unmounted and safe.

#### Nature of Request:  
Currently when shutting down the screen powers off pretty quickly in the shutdown process even as the system may still be writing data or etc.

I don't know how much Atmosphere could actually replace of the process, but if it could force a sync and maybe even go so far as powering off the card reader itself before the screen actually powers off it would help anyone who ever needs to remove the memory card for whatever reason.

### What component do you feel this would best fit within?
I honestly don't know which part this would apply to as an end user.  It sounds like Troposphere from the list given, but I'm just not sure.

### Why would this feature be useful?
This has been known to lead to problems with filesystem corruption happening more easily upon memory card removals especially.  I suspect this is also the reason I've had two memory card readers manage to fail -- each right after removing the memory card after a shutdown when it was working right up until the point of that shutdown.  While I do wait a while, it's impossible to know exactly how long I need to wait.  I guess sometimes it needs longer than others because I have definitely had issues several times.  I realize that most just use FTP or USB for transfers, but sometimes it's far more useful to actually remove the card (especially for me when actually going through the process of doing an update to Atmosphere itself.)  Not to mention really large transfers (especially when it's a high number of files rather than necessarily being a large size.  I ran into problems with this trying some RPGMaker games with EasyRPG-Player, some of which had thousands and thousands of files for some reason.  A few mods sometimes do like this as well.)

I don't know how realistic this is at this stage of the development process, but I do want to at least suggest it and hope it will be considered as something to be implemented at some stage of things even if it comes much later.

Masamune3210, 2020-12-02:
I'm not even sure how removing a card from the card reader would damage it unless you are doing it in a rush or forcefully or something? to the reader, nothing is different between removing a card when you are doing something with it vs when you aren't

SciresM, 2020-12-02:
not feasible/not in scope.

also, official hos fs sysmodule does do a sync/flush before shutdown.

closed 2020-12-02 by SciresM