Atmosphere-NX/Atmosphere #1224: boot2 flag presense in folder 010000000000bd00 causes a fatal error on boot

SoulBlast35, 2020-12-02:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
Upon booting, the Switch logo will appear for half a second before going to a fatal error
[ Describe the issue in detail and what you were doing beforehand. ]</br>
Simply Updating Atmosphere to 0.16
[ Did you make any changes related to Atmosphère itself? ]</br>
No
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?
Having "boot2" in the flags folder of 010000000000bd00 in "contents" causes it. Im not sure what program ID this is
[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>
All that was generated was a fatal error report, not a regular crash report

### System Firmware Version

11.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Official
Unsure about kips, however I know that I have MissionControl and sys-con installed, along with ftpd-lite
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

DevYukine, 2020-12-02:
This is an issue with `MissionControl` not with Atmosphere, you should remove it and wait for them to update for latest atmosphere changes. (for reference the boot2 file in the flags folder marks it to autostart on boot)

SoulBlast35, 2020-12-02:
Right, gotcha.

closed 2020-12-02 by SoulBlast35